supermerill/SuperSlicer #548: *Different* Retraction calibration test - distance vs. speed

TheAndr0id, 2020-10-08:
### Version
2.2.54.0

### Behavior
I like the new retraction test, it's quick, informative and it mostly just works.  The current test used temperature as one of the variables(a new tower for each temp), and while this does factor in, temperature is more important for other factors than stringing.  Could we get the same test with retraction speed as a variable?  Maybe lock the distance down and vary retraction speed over the column?

It's possible to do this all manually of course...

_Is this a new feature request?_
Yes

supermerill, 2020-10-08:
Is there an advantage to go at lower speed than the max?

qwewer0, 2020-10-08:
https://github.com/supermerill/SuperSlicer/issues/411#issuecomment-681146377 The article in the comment goes in detail with the retraction/deretraction speeds.

My explanation to the extreme ends.

- If we retract the filament too fast, then it could separate the solid filament from the melted, without moving the melted/half melted filament, leaving more than adequate amount of filament to ooze, because of thermal expansion.
- If we deretract too fast, then the melted filament can cause skipped extruder steps, because some materials (PLA) can increase its viscosity (non-newtonian fluid behavior).
- Obviously if we retract/deretract too slowly, it would take too long, and the nozzle would stay in place for too long, melting the printed part.

One more thing to remember for the retract/deretract speed in retraction calibration test, is that it should work with (marlin) firmware retraction.

I'm open for new information, so prove me wrong.

nhopfen, 2020-10-10:
The temperature variable has been very useful to fine tune materials where stringing is more temp related like PETG, It varies alot from one brand to another since it can have a wide range of formulations + it's goopy/blobby nature. As for speed, 60/20 has always given me good results with any material, PLA, PC, TPU, no exceptions. Of course these numbers are personal to me and needed to goof around to find them, so this could be valid, but I wouldn't come back to it as often. I love this new test!