supermerill/SuperSlicer #550: "Only one perimeter on Top serufaces" affects previous layers

Zulaner, 2020-10-09:
### Version
2.2.53 (Retraction calibration edition)

### Behavior
"Only one perimeter on Top surfaces": Unchecked
[image]

"Only one perimeter on Top surfaces": Checked
[image]
the arrow is to highlight this perimeter, that is not even part of the drawing

[Superman_Ear_Saver.zip.amf.zip](https://github.com/supermerill/SuperSlicer/files/5356773/Superman_Ear_Saver.zip.amf.zip)

neophyl, 2020-10-09:
On your 1 perimeter for top layers setting in the main configuration you have it set to Minimum Width 200%.  While that is great for sloping walls (and is pretty much what I use) in this case its causing the issue.  If you set it to 100-150% then it slices correctly.  Not sure why its doing that though.

supermerill, 2022-03-23:
can't solve

closed 2022-03-23 by supermerill