supermerill/SuperSlicer #551: Custom Start G-Code for Filament does not apply into the Gcode correctly

ymanyin, 2020-10-10:
### Version
v53 moar bugfixes

Version 2.2.53

### Operating system type + version
Windows

### 3D printer brand / version + firmware version (if known)
Voron v2.4

### Behavior
* _Describe the problem_
Custom Start G-Code should add my custom G-Code in the start. It does not do so.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_

Repoduce everytime it slices if you have anything typed in Custom Start G-Code for filament.

* _Expected Results_
Adding Gcode to the beginning part.

* _Actual Results_
Nothing is added. Only Star G-Code has problem. End G-Code is OKAY.

  * _Screenshots from __*SuperSlicer*__ preview are preferred_

[image]
[image]
[image]

_Is this a new feature request?_
No.

neophyl, 2020-10-10:
I tried the same value in the filament start gcode  area and it correctly inserted it into the output gcode when I sliced.  Then again I am using a cr10 profile with Marlin as gcode flavour.  As you haven't attached a saved project .3mf file (zipped up) we cant see what settings or options you have that may be effecting your output.  What gcode type have you selected ?  

Please just attach your project file and then there might be enough info to track down your problem.

ymanyin, 2020-10-10:
Thank you for your respond!
It is the default profile for Voron 2.4.
Here is the 3mf. I hope it helps!

[Flow Rate Test Print.zip](https://github.com/supermerill/SuperSlicer/files/5358605/Flow.Rate.Test.Print.zip)

supermerill, 2020-10-10:
This .3mf is a 2.2.54.0 one, not a 2.2.53.3
If you used the 2.2.54 (it's saying 2.2.53 because of a bug), it's an already reported bug. These are resolved in the latest nightly: https://github.com/supermerill/SuperSlicer/actions/runs/296381150

ymanyin, 2020-10-10:
My apology. It should be 2.2.54. I must have messed up something.
It should be the version with "monotonous" infill because I wanted to test it out.
It is good that it has already been reported. Sorry for duplicating.
Thank you for your attention and your great work!

boromyr, 2020-10-12:
I also encountered this problem. I had to add the startup script manually.
First experience in SuperSlicer, but I come from PrusaSlicer.
V: 2.2.54, for Windows 10

closed 2020-10-28 by supermerill