supermerill/SuperSlicer #552: Latest artifact "#493 separate bridge fan/speed for overhang from bridg..." not starting

1N4148, 2020-10-10:
Latest build (#493 separate bridge fan/speed for overhang from bridg...) won't start with existing config. Starting without config works. But after importing a config bundle SuperSlicers is closing and refusing to run.

supermerill, 2020-10-10:
erf.

1N4148, 2020-10-10:
When starting from scratch: Checking "Generate Supports" also terminates SuperSlicer.

supermerill, 2020-10-10:
I fixed the second bug, but I fail to reproduce the first one. Do you have a config / config bundle that makes it crash?

1N4148, 2020-10-10:
See the attached bundle. If I delete all profiles in the settings/print, SuperSlicer starts. But as soon as I try to import a backuped print-profile it terminates.

[SuperSlicer_config_bundle.zip](https://github.com/supermerill/SuperSlicer/files/5359745/SuperSlicer_config_bundle.zip)

1N4148, 2020-10-10:
PS: The "#533 #534 fix bad extruder initialisation (from 0d775d9ec829fd04" build works, but has the bug with the wrong temperature for the first layer

supermerill, 2020-10-10:
can't make it crash even with your bundle on an empty setting folder.
[image]

1N4148, 2020-10-10:
Strange. Can I provide anything more for debugging? Maybe cause of the "generate supports" in my configs? Then I'll wait for the next build where the second bug is fixed.

1N4148, 2020-10-10:
Did a quick test and changed the "support_material = 1" to "support_material = 0" in the print-configs. Now it works. So, first bug is caused of the second bug.

supermerill, 2020-10-10:
should be fixed in last nightly

1N4148, 2020-10-10:
Ok, will test it :)

1N4148, 2020-10-10:
Ok, works! Thx a lot! :)

closed 2020-10-28 by supermerill