supermerill/SuperSlicer #553: Auto speed and Top Layer with change at height

Zulaner, 2020-10-10:
### Version
2.2.53 (Retraction calibration edition)

### Behavior
when using height change modifying only the TOP Layer count to 1
the entire printing speed is affected
[image]
[image]

Zulaner, 2020-10-10:
[Heart_EKG_Ear_Saver.zip.amf.zip](https://github.com/supermerill/SuperSlicer/files/5359265/Heart_EKG_Ear_Saver.zip.amf.zip)

supermerill, 2020-10-10:
[image]

for layer range is from 0.6mm so it affect everything after 0.6mm

i don't really understand what's wrong

Zulaner, 2020-10-10:
See the speed...

supermerill, 2020-10-10:
[image]
Sure, the "short layer time" Colling option kicks int (see filament-> cooling, you can set 'enable fan if time below' and 'layer time goal' to 0 to check their effects)

supermerill, 2020-10-10:
Ok, that's the gapfill inside the 'rectilinear / monotonous with gapill' that makes the auto-speed go to very low speed (because of very thin extrusion).

closed 2020-10-28 by supermerill