supermerill/SuperSlicer #554: Overhangs - Threshold question/concerns

TheAndr0id, 2020-10-11:
### Version
2.2.53.3

### Operating system type + version
Windows 7 x64

### Behavior
This might just be an educational thing, but I'm seeing some weirdness with the "Detect Bridging Perimeters" option from Prusa Slicer (It is labeled as "Overhangs: On perimeters" in SuperSlicer.

From my understanding the default threshold value of 50% is supposed to duplicate the same setting as PrusaSlicer.  When I slice the same model with (as close as I can get) with the same settings in PrusaSlicer I get noticeable differences in the results.  The two jpgs below, the first is from PrusaSlicer, the second is from SuperSlicer.   The blue shows the overhangs in the writing, and it requires the threshold to be increased up to 80% (0.32mm for my 0.4mm nozzle) before the slicing looks similar.

So, is there something going on I don't understand, or is this something that needs to be looked at?  

(The model isn't mine, it's a tolerance test off Thingiverse - https://www.thingiverse.com/thing:4152967)

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[image]
[image]

supermerill, 2020-10-11:
prusa code:
`m_lower_slices_polygons = offset(*this->lower_slices, float(scale_(+nozzle_diameter/2)));`
superslicer code:
```
double offset_val = double(scale_(config->overhangs_width.get_abs_value(nozzle_diameter))) - (float)(ext_perimeter_width / 2);
        this->_lower_slices_bridge_flow = offset(*this->lower_slices, offset_val);
```

So if your extrusion width is 160% of your nozzle diameter, you may need a bit higher value to replicate.
Here the `- (float)(ext_perimeter_width / 2);` is here to say that the thing that count is the extrusion from the bottom, and the `config->overhangs_width.get_abs_value(nozzle_diameter)` is here to say that we want a % of the nozzle to be ouside of it to trigger the overhangs. The prusaslicer takes only the nozzle diameter into account.

But I may made a mistakes, it's easy to be confused on these things, too much operations and names.

TheAndr0id, 2020-10-12:
Now I am totally confused.   So the help tip for the threshold value ("As bridge" threshold) reads "The Minimum unsupported width for an extrusion to be considered an overhang.  Can be in mm or in a % of the nozzle diameter."

So with this set to the default of 50%, with a 0.4mm nozzle this should only trigger if 0.2mm or more is over the edge correct?  Now without knowing the context of the prusa code, that *seems* to be what they are doing.

I'm trying to follow your code (without context as well), SuperSlicer is using the overhangs_width (which I'm guessing is the "As bridge threshold" given the nozzle diameter.  Aka, using the 50% value against the nozzle width (0.4 in my/most cases) and then subtracting half the external perimeter width from that?

Then I'm guessing you are using that resulting value to set an extrusion as an overhang given that value.

Here's where I'm confused.  Given the above values with my external perimeter width set to 0.44mm,  then:
  0.50*0.40 - 0.44/2 = -0.02mm

Now when I set it to 80% this value becomes 0.1mm.

Now I'm assuming the offset function in SuperSlicer is different than the one in Prusaslicer.  If not, these values aren't the same.  I'm not 100%, but I think the values you are calculating are the mid point of the overhanging extrusion and prusa is using the half the nozzle width given the edge.  These values are using a different point of origin.  Unless the "offset" function has been updated, these aren't the same reference value.

I tried and failed to find the "offset" call in the code.  My C++ just isn't up to speed.  I looked through the change logs for src/libslic3r/PerimeterGenerator.cpp and saw that the half external perimeter was added in change  #414.  I didn't see an update to the offset function to compensate for this extra value.  It might have fixed the overhang cooling issue, but it looks to have caused issues with the bridging detection.

The underlying reason for this is this bridging perimeter detection is messing with auto support detection.  I've printed this model sliced both in Prusa and Super and it prints perfectly without issue.

supermerill, 2020-10-12:
The offset function is the same, it offsets the polygon we work on.

You have the contour of the previous layer. You offset it by half nozzle for the prusa code.

First, I just replaced `nozzle_diameter / 2` by `config->overhangs_width.get_abs_value(nozzle_diameter)`, so 50% should be the same.

but then, I had this report: 
_Originally posted by @Pavulon87 in https://github.com/supermerill/SuperSlicer/issues/414#issuecomment-677646970_

So I changed my code to try to answer the desirable result, but I may have forgotten to change the default value...

TheAndr0id, 2020-10-12:
Then I think I see the problem.  If offset function is expecting a positive value in terms of nozzle size / 2, then the default of 50%(for 0.4 nozzles anyway) is passing a negative value.  The `- (float)(ext_perimeter_width / 2)` is causing **all** overhanging perimeters to be treated as a bridge.

In order to get this back to what Prusaslicer uses I'll need to set this manually to 0.42mm (half nozzle width + half external perimeter).  At a minimum the tooltip needs to be updated because what it says the value does doesn't...

I don't think this is the solution to issue #414

closed 2021-06-04 by supermerill