supermerill/SuperSlicer #556: Feature Request: Bed Exclusion Zones

digitalentropy, 2020-10-11:
I don't know how difficult it would be to implement, but being able to define a permanent or per-plate "bed exclusion zone" would be an *excellent* feature.

Trying to manually adjust parts to not print over bad or damaged areas of the print bed is often somewhat inconsistent, and I don't always remember to do it.

There's an open request in PrusaSlicer, but knowing them it's unlikely they'll implement it anytime soon: https://github.com/prusa3d/PrusaSlicer/issues/3866

Makhaira, 2020-10-26:
A low tech solution would be just colouring in part of the texture file you load for that bed to show the bad spots.

digitalentropy, 2020-10-26:
It’s an okay hack, but really inconvenient and imprecise, and wouldn’t allow for use of automatic positioning.

supermerill, 2020-10-27:
Maybe I can let you add "fake objects" that will be saved in the profile. If they can't be moved by the arrange button, that should do the trick.

digitalentropy, 2020-10-27:
That could work!

Can the exclusion object texture be markedly different so it’s obvious it’s not a real object?

supermerill, 2020-10-27:
It should, but it's just an idea, I have a plethora of feature to add.

digitalentropy, 2020-10-27:
Of course. Have to prioritize.

sprior, 2022-12-26:
In case you weren't aware, Bambu Studio implements "Excluded bed area" in their "Printable space" section of the machine configuration.  This seems to correspond to the General -> Size and coordinates section of the Printer settings in SS.  Bambu Studio source is at https://github.com/bambulab/BambuStudio