supermerill/SuperSlicer #557: Feature Request: Strength Dots

ricktm, 2020-10-12:
I really like the current ironing support, and would like to recommend a new infill feature that would build on it, an idea I'm calling Strength Dots.  Or spot welding. To implement this, a nozzle would be lowered to iron not only a top layer, but every infill layer in a polkadot like pattern, with the print nozzle pressing down with a preset amount on each dot point, into a print. This would create a lego-like pattern which would cause better fusion to each layer.

### Version
_Version of SupserSlicer used goes here_

_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

supermerill, 2020-10-12:
the problem is that each infill is dependent on the previous layer and on the next one.
You have to punch over previous holes, and let hole for the next layer, unless there's a top one...
It's complicated to implement and I don't have solid data to enforce the "better fusion to each layer" claim.

ricktm, 2020-10-12:
Two ways to work around your cited problems, 

One is thru alternating dots for each layer, either thru positioning of the welds or alternating dot layers. I believe alternating dot position may be the better solution.

or Two, while printing dots instead of pressing down on the nozzle, lift the nozzle (except on top layers), this way a proceeding layer would have more surface area to to fuse to.

Im sure sure many more ways to implement this, the main goal however is to create better layer fusion. much like how knitting can increase the strength of fabric.

mocksoul, 2020-11-24:
u can just make line width 150%+ of your nozzle size (e.g. 0.5+ with 0.4 nozzle). That will make print much more rigid, but more difficult to print overhangs without artefacts (good cooling needed)