supermerill/SuperSlicer #558: Feature request: Idea cloud

Coffee0297, 2020-10-12:
Can we get support to upload direct to idea cloud? its soo nice and makes so you don't have to mess with octorpint's bad UI

Kind regards Tonny

supermerill, 2020-10-12:
I need more informations and links to how to do that.

Coffee0297, 2020-10-12:
https://github.com/Raise3D/OctoPrint-Raisecloud
thats the octo plugin, i can't seem to find how to manually upload tho

supermerill, 2020-10-12:
Isn't this to add an octoprint printer to the Raisecloud?

> i can't seem to find how to manually upload tho

you can't manually upload to raisecloud?

You have to explain me how to use or upload to 'raisecloud'.
It doesn't have an api?

Matts-Hub, 2020-10-15:
According to this thread there isn't currently any api to upload to the raise cloud https://forum.raise3d.com/viewtopic.php?t=20763

supermerill, 2020-10-15:
!remindme 1 year

closed 2021-04-27 by supermerill