supermerill/SuperSlicer #559: [FEATURE REQUEST] Selectable speed setting for where different materials interface in Z

ogland, 2020-10-12:
Basically what the title says. The adhesion with some soluble support materials is really bad, and it would be nice to be able to print supports fast with the normal material, except for when it interfaces with the soluble material. As it stands now, supports has to be printed at about 15 mm/s to properly adhere. Some kind of extrusion multiplier for interfacing layers might also help with the results, at least for the supports.

supermerill, 2020-10-12:
* There is already a support speed and a support interface speed, what is exactly the problem if you set them to 60 & 15 ?
* fan for parking head: I can add a "parked" in the fan speed section of the filament (if the printer is really multi-extruder). The gcode needed will depends on the firmware... 
* A full raft below the support-raft? (supports need their raft or they will fall even easier than actually)

ogland, 2020-10-12:
> * There is already a support speed and a support interface speed, what is exactly the problem if you set them to 60 & 15 ?

Yes, but the issue is that with soluble supports, I need to set the support and interface speed to 15 mm/s for the different filaments to adhere to each other. If they could have a speed for when they interfaced with each other, I could set the speeds much higher for the rest of the layers where they only adhere to themselves.

> * fan for parking head: I can add a "parked" in the fan speed section of the filament (if the printer is really multi-extruder). The gcode needed will depends on the firmware...

Fantastic! Also if the gcode output could select the proper fan (P0 & P1) for the active toolhead, that would be even better.

> * A full raft below the support-raft? (supports need their raft or they will fall even easier than actually)

A support raft only. The issue here is that the support raft is printed with the soluble support toolhead and I can't change that without also changing the toolhead for interface layers.

supermerill, 2020-10-12:
1) 
> If they could have a speed for when they interfaced with each other
isn't that the interface speed?
> , I could set the speeds much higher for the rest of the layers where they only adhere to themselves.
isn't that the default support speed?
I'm not using interface that much, and i don't have a multi-tool printer, so you have to en-light me a bit on that.

3)
I'm assuming you selected the setting "soluble on interface only" ?
So what you want is to not use the soluble on interface that touch the build plate, right?

ogland, 2020-10-12:
> > If they could have a speed for when they interfaced with each other
> 
> isn't that the interface speed?
> 

The interface may be up to three layers and it's only the bottom of those ones that needs to go slow. 

The first support layer is printed on a raft of soluble supports so that needs to go slow.

The layer printed on the interface layer also needs to go slow. 

> > , I could set the speeds much higher for the rest of the layers where they only adhere to themselves.
> 
> isn't that the default support speed?
> I'm not using interface that much, and i don't have a multi-tool printer, so you have to en-light me a bit on that.
> 
> 

Yes and no. I answered this partially in the previous question. 

> I'm assuming you selected the setting "soluble on interface only" ?

Yes

> So what you want is to not use the soluble on interface that touch the build plate, right?

I do want soluble there, but not the entire bottom layer underneath the supports, such as it is now.

supermerill, 2020-10-12:
Can you send me a little project file where the problem occurs?

ogland, 2020-10-12:
Sure. Here it is.

[Shape-Sphere.zip](https://github.com/supermerill/SuperSlicer/files/5365947/Shape-Sphere.zip)

ogland, 2020-10-12:
Here's the rendered gcode file.

[Shape-Sphere_0.2mm_PETG_MK2.5.zip](https://github.com/supermerill/SuperSlicer/files/5365949/Shape-Sphere_0.2mm_PETG_MK2.5.zip)

supermerill, 2020-10-12:
1)
It is that ?
[image]

3)
support-raft should not be printed with interface extruder if "soluble on interface only"
[image]

ogland, 2020-10-12:
Yes! Just like that, but also the layers on the actual object that interfaces with the interface/soluble material also needs to have a speed setting.

supermerill, 2020-10-12:
I can add an "support interface layer speed" in the filament setting. It will trigger when you print over an other filament.

Note that it's an idea, but implementing it may be a bit long, as it's not easy to work with this kind of stuff.
I have to make "speed" modifier work at the gcode level instead of the surface level, so also a bit of difficult preliminary work.

ogland, 2020-10-12:
Nothing is expected, everything is appreciated. The fan issue is also pretty important. Should I make a bug report or a feature request separately for that or is this enough?

supermerill, 2020-10-12:
it's better if it's separate, as I can tag it resolved.

ogland, 2020-10-21:
Ok so a small update on a detail. The support material raft material error was 100% because of my botched starting gcode.