supermerill/SuperSlicer #560: envoie gcode à mon imprimante

harcrafri94, 2020-10-12:
### Version
_Version 2.2.53.2 windows 64

Bonjour
sur superslicer j ai que la proposition d'envoie via un hôte http quand je fait mon GCODE sa me proposer bien d'envoyer à mon imprimante vis le cable USB mais après j ai que des messages d erreurs je peu pas faire via le cable du coup je doit mettre en carte SD
merci à vous :)

supermerill, 2020-10-12:
Je me suis permis d'éditer le message, car il était complètement incompréhensible.

Il y a deux façons principale d'envoyer le gcode à l’imprimante. Via un fichier (donc via une carte sd), ou via un serveur 
d'impression (octoprint, principalement). Si tu as des problème pour configurer octoprint, regarde là https://help.prusa3d.com/fr/article/envoyer-des-fichiers-a-octoprint-duet_5507

Il a existé une troisième manière qui consistait à utiliser le slicer en tant que serveur d'impression, avec l’imprimante connecté au pc via un cable usb. Je ne sais pas si cette fonction est toujours active. Elle a un gros défaut qui est d'interrompre l'impression si l'ordinateur se met en veille ou redémarre (et il est fréquent que windows fasse cela par lui-même).

There is two way for sending the gcode to the printer: by writing a gcode file (and then using a sdcard or a usb stick), or by sending the gcode to a printer server (octoprint). If you ahve some trouble to setting up the octoprint, look here https://help.prusa3d.com/fr/article/envoyer-des-fichiers-a-octoprint-duet_5507

There were a third way, where the slicer is the printer server, connected to the printer via a usb cable, but it's not recommended as any problem on the host computer will stop the print for good.

harcrafri94, 2020-10-14:
Bonjour 
j aimerais le faire via cable usb pas octoprint