supermerill/SuperSlicer #561: [BUG?] Correct fan isn't selected with P0/P1..

ogland, 2020-10-12:
### Version
Any? 

### Operating system type + version
Any? 

### 3D printer brand / version + firmware version (if known)
Marlin 2.0.6

### Behavior
* _Describe the problem_
When printing with IDEX and different filament types, different part cooling settings is needed, however the slicer doesn't output the P0 parameter for T0 or P1 for T2 etc when sending M106. 
* _Steps needed to reproduce the problem_

Slice with two different filament profiles for two different extruders. 

* _Expected Results_

Setting proper fan speed at tool change with P designation. 

* _Actual Results_

Erratic fan behavior. Sometimes the T1 fan spins when the T0 fan should spin and T1 filament profile has fan set to 0 or something else.

supermerill, 2020-11-15:
can you give me a project file to reproduce?

ogland, 2020-11-15:
@supermerill I can do that sometime soon! What I noticed is that the fan speed setting with M106 happens before the tool change gcode, and so the wrong fan gets targeted.

closed 2021-08-22 by supermerill