supermerill/SuperSlicer #565: Bridge auto angle: FAIL

Zulaner, 2020-10-13:
### Version
2.2.54 (night build)

### Operating system type + version
Windows

### Behavior
the automatic angles are wrong

#### Project File (.3MF) where problem occurs
[Bridging_Test.zip.amf.zip](https://github.com/supermerill/SuperSlicer/files/5373419/Bridging_Test.zip.amf.zip)

Zulaner, 2020-10-13:
[image]

neophyl, 2020-10-13:
Actually the one on the right would be my preferred printing angle for the bridge as it’s the shortest distance. The ok one looks very long so that will droop a lot.

Zulaner, 2020-10-13:
the perimeters being detected as overhang, these should not be considered for calculating the bridge distance

supermerill, 2020-10-13:
Currently, the brute-force algorithm check every 2° angle, so it's easy to not be optimal if the optimal angle isn't a multiple of 2.

also duplicate of #234 #149

It's something to revamp...

girrrrrrr2, 2020-10-15:
> Maybe i could publish the hidden setting in the meantime?

That would probably an easy temporary fix. Maybe have a screen somewhere that has a bunch of settings like this, with the request to reach out if you find a good parameter, or something.

closed 2021-06-04 by supermerill