supermerill/SuperSlicer #566: Typos in Retraction Calibration Screen

girrrrrrr2, 2020-10-14:
### Version
2.2.54

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Unnecessary

### Behavior
Extruder Retraction calibration has a typo and the title says Flow Calibration
Typo - Multiply it by your step parameter (0.2, 0.2, 0.5 or 1)
Should be - Multiply it by your step parameter (0.1, 0.2, 0.5 or 1)

Title Box Says - SuperSlicer - Flow Calibration
Should be - Super Slicer - Retraction Calibration

Super simple issues, nothing really that crazy, just thought id point them out so that you knew.

supermerill, 2020-10-14:
>Multiply it by your step parameter (0.2, 0.2, 0.5 or 1)
>Multiply it by your step parameter (0.1, 0.2, 0.5 or 1)

can you put in bold the characters that are wrong? I can't spot them.

digitalentropy, 2020-10-14:
Multiply it by your step parameter (**0.2**, 0.2, 0.5 or 1)

girrrrrrr2, 2020-10-14:
> Multiply it by your step parameter (**0.2**, 0.2, 0.5 or 1)

Yes this. 0.2 is listed twice.

closed 2020-10-28 by supermerill