supermerill/SuperSlicer #567: Enabling "Generate support material" checkbox instantly crashes SuperSlicer

digitalentropy, 2020-10-15:
### Version
Nightly build from commit 8431ca1

### Operating system type + version
Windows 10 X64

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK2.5S

### Behavior
Attempting to enable support material generation on the print settings tab seems to result in SuperSlicer immediately crashing. I've attempted to fix this by deleting the entire profile directory and re-adding profiles and printers, but the same behavior is observed. I thought it might be profile related, but it appears to happen on factory profiles as well.

Nothing specific is needed for me to reproduce it. All I have to do is start SuperSlicer, select the Print Settings tab, and attempt to enable support generation. The application immediately crashes. It doesn't seem to matter if I have a model already added or not.

supermerill, 2020-10-15:
already reported in #552, have to check why it wasn't fixed.

digitalentropy, 2020-10-15:
The error appears to be mine. I thought I was on the absolute latest nightly but I was one commit off. Once re-downloaded the latest nightly the issue appears to have been resolved. Closing for now.

closed 2020-10-15 by digitalentropy

supermerill, 2020-10-15:
sometimes, I erase the latest commit to keep commits as consistent as possible, for ease of cherry-picking and merging.