supermerill/SuperSlicer #568: Feature Request: Add ability to start/end perimeters inside the model to hide seams

digitalentropy, 2020-10-15:
If there was a feature to start the beginning and ending point of an outer perimeter inside the part, that should hide the seam much better. Wiping inward can only do so much.

There's a decent explanation of what I'm talking about here: https://forum.simplify3d.com/viewtopic.php?f=23&t=3567&p=15604&hilit=seam+reduction#p15604

The distance from exterior for start/end could be configurable, as could the length of external perimeter gap.

supermerill, 2020-10-15:
It's already the case for the end point.
I agree for the starting one.

Matts-Hub, 2020-10-15:
This is a great idea, as far as I know it was turned down as a [feature request](https://github.com/slic3r/Slic3r/issues/3299) for slic3r because of the [patent](https://patents.google.com/patent/US20130095303) that stratasys has on it. I wrote an article that features it [here](https://mattshub.com/blogs/blog/layer-seams). I'm not sure what this means regarding its implementation in superslicer.. I guess thats up to @supermerill :p

digitalentropy, 2020-10-15:
I believe the best way to handle this is to expose various variables regarding the start/end point of external perimeters to the UI. The default behavior can still be as it is now, but the user would then be able to tweak various settings for start/end position, distance inside or outside the perimeter, and incidence angle of the extrusion line.

digitalentropy, 2021-01-07:
Any chance of revisiting this now that major merging has been done?

Kosh42, 2021-05-31:
This would be an awesome feature.

digitalentropy, 2022-01-02:
> I believe the best way to handle this is to expose various variables regarding the start/end point of external perimeters to the UI. The default behavior can still be as it is now, but the user [...]

Any odds of getting this implemented as a test?

supermerill, 2022-01-04:
I have a request to do the opposite: to not end inside perimeters
#2156

digitalentropy, 2022-01-04:
Hmmm... considering this is one of the best ways to deal with certain types of artifacts, would the adjustable method described above solve both problems?

digitalentropy, 2022-04-28:
Is this getting close to implementation with some of the changes I saw for little wipe with external perimeters?

supermerill, 2022-04-28:
check the latest nightly and tell me if it's enough.

supermerill, 2024-09-25:
now there is a startwipe, should be enough.
reopen if it's still not satisfactory

closed 2024-09-25 by supermerill