supermerill/SuperSlicer #573: Bridge improvement

boromyr, 2020-10-16:
### Idea
I had a simple idea to improve the printing of the decks, preventing them from sagging, but I'm not sure of its effectiveness. It is more a hypothesis that I share with the community for opinions from more experienced people.

The printing of a bridge begins and ends in its joint points, being printed on the XY plane until the second joint is completed, but in the meantime, the filament is inevitably lowered on the Z plane, especially if it is long and the width of extrusion is great. Then it would be possible to add a correction factor before the second joint, which raises the Z-axis slightly before touching the perimeter, and then immediately returns to the current layer, before the casting cools, a bit like stretching a chord by making an angle on it. a wall instead of directly.

Formalizing the idea, what is created is a right triangle with a ratio of the sides 1/1000 (as an example), and the hypotenuse is the print filament, which will eventually coincide with the lower cathetus once that sin (a) = 0.
I think it is enough to use the same function of "raise Z" of the retraction, and call it at the end of the bridge; a small final acceleration would be of great help.

#### pro
* linear bridges instead of slightly concave
* the average length of suspension bridges increases
* saving of previously needed media
* better quality

#### versus
* may not work as expected
* the joint may not adhere well if you do not work quickly
* may not be suitable for series bridges, for example, those overfill

If it works it could be successfully integrated into SuperSlicer.

yschroeder, 2020-10-19:
I think what happens with long bridges is that they sag and cool during printing. You will already have a solidified curved extrusion when you would try to stretch it to the anchor point. Also I think you cannot really stretch the filament the way you want, as it is soft and would just deform right at the nozzle.

But as we are discussing this another idea came to my mind: Would it be possible to move the nozzle on a curve away from the print surface to counteract the sagging of the filament? Lets say I know that a long bridge sags 1mm in the middle, would it be possible to extrude a curve that is 1mm higher in the middle and thus cancels out the sagging?

supermerill, 2020-10-19:
Don't you both had the same idea?

yschroeder, 2020-10-19:
I understood @boromyr that he just wants to nudge up the nozzle slightly before connecting the second anchor to pull up bridge.

But in case we had the same idea it must be a good one 😄

boromyr, 2020-10-19:
Tomorrow I will make a draft of the drawing in which I outline the idea well. It is difficult to be able to convey a graphic idea in words, and it is easy not to understand each other. A pattern is worth a thousand words! If you can do one yourself, it can be proven that combining 2 ideas is even better 1!

robthide37, 2020-10-30:
I think prob may be that if you arc the print the line will be longer and then actually be counter intuitive