supermerill/SuperSlicer #578: Bridge/Overhang Fan Speed

xFuryx87, 2020-10-18:
### Version
2.2.54 nightly

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Anycubic I3 Mega S, Marlin 2.0.5.4

### Behavior
The fan speed is not only applied to bridges and overhangs, which results in unsightly perimeters when printing with PETG (red layers in the screenshots)
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]
[image]

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

supermerill, 2020-10-18:
can you post your project file?

xFuryx87, 2020-10-19:
> can you post your project file?

I will post it as soon as I'm home again.

xFuryx87, 2020-10-19:
> can you post your project file?
Done!

supermerill, 2020-10-19:
printer setting -> fan startup delay, put 0.

xFuryx87, 2020-10-20:
> printer setting -> fan startup delay, put 0. It's a setting to let some time for the fan to start. Or it won't even has a chance to ramp up its speed.

worked! you are my hero! thank you very much

closed 2021-06-04 by supermerill