supermerill/SuperSlicer #579: Control when bridge infill is applied based on infill density and extrusion width

jschuh, 2020-10-18:
Any interest in me porting prusa3d/PrusaSlicer#4374 to SuperSlicer? It adds an area threshold setting to determine when bridging flow should be used over infill. I've been running it on my custom build of Prusa Slicer for a very long time now, and my experience is faster print times (I use slower bridging speeds) and better solid layer adhesion over infill, with no visible reduction in print quality.

I've been playing with SuperSlicer for the last few days and hadn't found an equivalent way to do this (although I may have missed it).

supermerill, 2020-10-19:
what's the use-case?
If you don't use a bridge flow over a sparse infill, your top layer will drop even more, and the over-bridge flow may have a harder time to recover.

jschuh, 2020-10-19:
Fair question. The point of having a threshold setting is to use the bridge infill flow precisely where it's needed—so that you get bridging flow over sparse infill, but not over dense infill where it slows down the print and can cause adhesion problems (see prusa3d/PrusaSlicer#3814). So, I guess another way to phrase this change is: _Don't use bridging flow over dense—but not necessarily solid—infill._

As an example, take the case where you're using 99% infill because you want effectively solid infill, but need some leeway to account for an imperfect extrusion multiplier or looser filament tolerances. As the code is currently implemented, the first solid layer over that 99% infill is always going to use bridging flow, which means slower speeds from the bridging flow and poor layer adhesion due to heavy cooling bouncing off the dense infill. Hence why I think it makes sense to have a configurable threshold for what counts as sparse versus dense infill.

Accepting the above, I haven't stress tested the proposed default threshold with a combination of ironing, very thin layer heights, and minimal top layers. So, it's certainly possible that I should consider a more conservative default value. Do you have suggested settings that I could use to help determine a safer default?

supermerill, 2020-10-19:
i'm ok with the premise: don't use bridge flow on 90% infill, but I won't agree with the code.

1) you should cut the surface to have bridge over sparse infill and solid over "almost solid"
2) "almost solid" is imo when the bridge hasn't enough volume to be extruded: 
  * check available volume: surface.area() * layer.height
  * compute bridge volume: bridge_flow.mm3permm * surface.area() / bridge_flow.spacing()
(note: use scaled coordinates everywhere)
Then, you can calculate how many plastic has go go inside the infill: needed for bridge - available 
And then, by comparing / multiplying by the infill% (to see if there are enough holes for the overflow) and the total flow (to see the % of overflow, useful for thin layers on big nozzles) and decide (via the setting) if the bridge flow can be kept.

basically, what the comment below said, but it's a lie currently.

Put the setting (%?) to 0 is equivalent to disable "bridge infill".

though?

Note: if you want to merge something, merge it against dev branch.