supermerill/SuperSlicer #580: first layer - extra perimeter

rkoehler2017, 2020-10-19:
latest 64Bit SS
Windows 64

Maybe it's a wrong setting but SS slices me an extra perimeter on the first level,
0.8mm Nozzle, 0.4 Layer, no brim
[image]

supermerill, 2020-10-19:
your first layer compensation is positive instead of negative.

rkoehler2017, 2020-10-26:
i checked all settings in SuperSlicer but i can not find - first layer compensation

[FilaTrenner v1_PLA.zip](https://github.com/supermerill/SuperSlicer/files/5439052/FilaTrenner.v1_PLA.zip)

supermerill, 2020-10-26:
in print->slicing->modifying slices->XY compensation->First layer

rkoehler2017, 2020-10-26:
Great, Thanks, 
that was the clue

closed 2021-06-14 by supermerill