supermerill/SuperSlicer #581: First layer temperature only on skirt

bigsubgit, 2020-10-19:
### Version
2.2.54

### Operating system type + version
MacOS Catalina

### 3D printer brand / version + firmware version (if known)
Custom printer with RRF, but I have the same problem also with another printer, with Klipper on board.

### Behavior
I noticed that first_layer_temperature is applied (now, but I don't know when start this behavior) only on the skirt, then all the first layer is printed at temperature_layer, and so on the rest of the print.
I think it's a mistake to setting new temperature after skirt, or not?

[image]

[image]

[image]

What could I check?
Regards.

bigsubgit, 2020-10-19:
Here is more visible the temperature of first layer

[image]

supermerill, 2020-10-19:
Thanks for the report.
But already reported fixed, and warned about in the release page. download the nighlty, or wait some days to let me correct other bugs.

closed 2020-10-19 by supermerill

bigsubgit, 2020-10-19:
@supermerill many thanks and sorry if didn't check before.