supermerill/SuperSlicer #585: Gap fill causing excessive travel

harikattar, 2020-10-19:
### Version
2.2.54 (nightly 2020-10-17)
Same results in 2.2.53.2

### Operating system type + version
Linux appimage

### 3D printer brand / version + firmware version (if known)
n/a

### Behavior
When using one of the (filled) variants of infill patterns, excessive travel moves are generated.

Examining the first layer when using concentric (filled), it travels to a completely different area of the print every time a gap fill is generated.

[image]

To reproduce, slice with concentric (filled) and examine the first layer with travel.

Without gap fill (just concentric), the infill finishes each area of the print before moving on to another.

[image]

The intersection between circle and square perimeters generates an extreme number of gaps for concentric that demonstrates this issue.

#### Project File (.3MF) where problem occurs
[fan test.zip](https://github.com/supermerill/SuperSlicer/files/5403921/fan.test.zip)

supermerill, 2020-10-19:
mmm, i think I understand where this problem comes from.
Not that easy to fix, but manageable.

mwbrady, 2022-07-05:
Is there any fix for this in the works? I have a very similar model that has given me excessive travels on gap fill, but I've been getting increasingly annoyed with the seemingly illogical move and travel planning implemented by superslicer. I know it's complex, but Cura 5.0 is leaps ahead on this. For thin parts that are mostly walls, thin walls and gap fill, I often find that the paths on a single layer in superslicer will be travelled nearly twice by the nozzle because the travels and backtracking become so excessive. I've been using Superslicer for a couple years now and I'm pretty well versed in the settings. Nothing I've tried will fix it. Other slicers like Cura 5 and Simplify3d don't have this issue. The paths in that seem to minimize the travels and total movement of the nozzle in a layer. I still think Superslicer it is a better slicer overall but the move planner has definitely fallen behind and for those trying to reduce print time, every extra move counts. I can provide examples of Superslicer vs Cura if necessary.

Version 2.3.57.9 on Windows 11

My model's first layer pictured with excessive travel between gap fill.
[image]