supermerill/SuperSlicer #587: Adaptive Support Cubic infill for supports (again)

espr14, 2020-10-19:
The infill is in PrusaSlicer but cannot be used in support_material_pattern.
#582

neophyl, 2020-10-19:
Did you not get that when Super Slicer gets merged into a proper release of Prusa Slicer that it will then be available ?  Basically you will just have to wait until a future release.

supermerill, 2020-10-19:
on the dev branch?

espr14, 2020-10-20:
Previously, PrusaSlicer was merged into SuperSlicer. Not SS into PS.
However, I am missing a feature in PrusaSlicer and SuperSlicer as well. Therefore, their merge will not bring that feature. The requested feature is availability of adaptive cubic infill (or even all infills) in the support_material_pattern.

neophyl, 2020-10-20:
I've currently got PS 2.3.0 alpha 1 installed and the adaptive cubic infill is available in the infill list in PS.  I haven't played with it yet to see exactly what it does different but its there.  Along with Support Cubic.

So I would expect when Super Slicer does do a merge from their proper release that it will appear.  Seems to be the way it works.

supermerill, 2020-10-20:
> PrusaSlicer was merged into SuperSlicer. Not SS into PS.

?

the supermerill/dev branch is my merge of ps-ss

espr14, 2020-10-20:
Any infill can be used as a support infill. 2D infills without any change, 3D infills (cubic, gyroid etc.) require sheath.

It's not there:
[image]