supermerill/SuperSlicer #589: higher speed on external wall bridge than set

Coffee0297, 2020-10-20:
### Version
2.2.54

### Operating system type + version
Win 10

### Behavior
When slicing this model i get a higher speed on external wall when there is a bridge. Not sure where it gets that speed from bc its not a different feature
[image]
[image]

#### Project File (.3MF) where problem occurs
[Test cube 3030.zip](https://github.com/supermerill/SuperSlicer/files/5409401/Test.cube.3030.zip)

supermerill, 2020-10-21:
that's your small_priemter_speed kick's in, set it to 50%, same as external.

Coffee0297, 2020-10-22:
Allright, that makes sense, and i see you made it a label in the legend 👍 thanks!

closed 2020-10-22 by Coffee0297