supermerill/SuperSlicer #590: [FEATURE REQUEST] Fan speed for support's

Zulaner, 2020-10-20:
Basically, it is to create the functionality to define the fan speed for the supports

supermerill, 2020-10-20:
Is it really needed? supports are over other supports and shouldn't need any special fan.

Zulaner, 2020-10-20:
I'm printing TPU at LH 0.2 with 40% fan.  the bottom support interface is welded to the piece even with 0.3mm distance
if I had the possibility to put more speed on the fan, I think it would be successful

Zulaner, 2020-10-20:
[image]
[image]
[GOPROCLOUD.zip.amf.zip](https://github.com/supermerill/SuperSlicer/files/5410275/GOPROCLOUD.zip.amf.zip)

supermerill, 2020-10-20:
So you don't want a higher fan for support but for the last support interface layer?
Can you try with a simple test piece (a simple T or L piece with big 90° overhang), with your best config and the same code but with a` M106 255` added just before the last support layer will be printed ?

Zulaner, 2020-10-20:
the problem is in the first support interface, not the last.
I want higher fan for all support's not on the last
[image]

another interesting feature was the possibility to choose whereinterface (floor/roof)
this feature could also solve this situation by creating an interface with greater spacing
[image]

Zulaner, 2020-10-20:
:)
[image]

Zulaner, 2020-10-20:
[image]
[image]

Makhaira, 2020-10-27:
Have you been able to confirm this works by manually editing the gcode @Zulaner

rhyonly, 2020-10-31:
This is pretty weird!! I 100% sure my support used to have a different fan speed than the rest of the part (100% for support compared to 75% for the part), It overcooled the support, make it very brittle (for PLA) that removing support is much easier. I accidentally delete the profile and I can't get it back 😭😭.  Is this actually not a feature in SuperSlicer? but I used to used them all the time, if there is a setting for what I mentioned, what is it?

Zulaner, 2020-11-18:
FAN Speed Default: 30%
Support: Interface layer  3
[TPUSupport_FAN.zip](https://github.com/supermerill/SuperSlicer/files/5560398/TPUSupport_FAN.zip)

With FAN 100% (botton: support interface and top: layer above support interface
[image]
[image]
[image]

Without FAN
[image]
[image]
[image]

Zulaner, 2020-11-18:
I would say it works
and if have the possibility to define the interface type in the botton and top 
separated, it was amazing :)

supermerill, 2022-03-23:
#2326