supermerill/SuperSlicer #592: Struggling to Slice Model with Thickness = Nozzle Increment

CCS86, 2020-10-21:
2.2.53 / Win10 64 / Ultimaker + Duet Maestro

When I design models, I generally work to make sure wall thicknesses (on the XY plane) are an even increment of nozzle diameter, (ie 0.4 x 4 = 1.6mm). SuperSlicer seems to struggle with this though, and leaves a gap between inner perimeters.

In this case I have a model with a 1.6mm wall, so 2 outer and 2 inner perimeters should fit perfectly. Zooming in on a single layer, there is a visible gap in the middle. Using calipers, I estimate this gap to be about 0.05mm. In theory changing the width setting from 0.4mm to 0.41mm (for all line types) should almost completely close this gap. However, I see no change in the sliced print with widths set to 0.4, 0.405, 0.41, 0.42, 0.45mm. Identical:

[image]

Is there a setting I am missing, or is this a bug? Even if I set it to 0.8mm widths, I get the identical 4 lines, not 2. It's like the width settings are completely nonfunctional.

[Test.3mf.txt](https://github.com/supermerill/SuperSlicer/files/5416004/Test.3mf.txt)

CCS86, 2020-10-21:
Changing the nozzle diameter in the printer settings causes the expected behavior. So I'm left with 2 questions:

*Are the width settings inactive"

*Why is SuperSlicer overlapping perimeters by around 0.013mm?

supermerill, 2020-10-21:
[image]

here you have a little help info. With your layer height, external perimeter width & perimeter width, you should have at least 0.81 mm for your wall thickness. If you make your model with 1mm wall thickness, you may want to increase your min to ~0.98mm, to do so increase your perimeter width by ~0.08mm.

CCS86, 2020-10-21:
Okay, I guess Cura handles that flow math differently. It must alter the flow rate to make a 0.4mm line width complete without voids, because every time I ask it for a 0.4mm line, that is what I get.

Does it really make sense to command a 0.4mm line width, but have the slicer make the lines 0.387mm instead?

Regardless of that, the line width setting in SuperSlicer appears non functional. If it is locking the outer and inner perimeters together, with a certain amount of overlap, but the inner perimeters don't touch each other, clearly I should be able to command a thicker line width and close that gap. This behavior works when changing the nozzle size setting, but nothing happens for commanded line width changes. I feel this is a bug.

I even tried updating my model to a 1.63mm wall thickness instead of 1.60mm, but the behavior is exactly the same.

supermerill, 2020-10-21:
The threshold to convert perimeter to gap fill is maybe a bit too agressive...
I'll adjust it when i'll revamp the perimeter generator.

imo, it's better to have a little gap fill between perimeters, so they won't push each other initially.

CCS86, 2020-10-21:
I really don't think the new title is accurate. 

I don't see how this has to do with gap fill. The gap is so small (only about 12% of nozzle width), that it is not getting any gap filling applied, which is good, because this wouldn't work well any way.

I see two issues

1. The line width settings appear non-functional

2. For a commanded line width, SuperSlicer generates a smaller line width to avoid voids from non-rectangular extrusion profile. I would argue that the slicer should generate line widths the exact width defined by the user, and use commanded extrusion volume to ensure no voids. This one is more a matter of philosophy though, where #1 seems to be an objective bug.

supermerill, 2020-10-21:
There is no "line width".
Read tooltips and try modifying things.
I don't have time to try to convince you that these settings are functional.

CCS86, 2020-10-21:
I considered "extrusion width" to be synonymous with "line width". Sorry if that was unclear.

As shown in my original posted screenshots, varying all the "extrusion width" settings had no impact on the sliced print.

CCS86, 2020-10-21:
Maybe this will illustrate better:

[image]

[image]

[image]

[image]

supermerill, 2020-10-21:
[image]

CCS86, 2020-10-21:
I read the tool tips and am not sure why you are confused here. 

"Set this to a non-zero value to all a manual extrusion width"

Pretty simple. I have specified a non-zero value there, and SS is ignoring it. The screenshots above show that for 3 different commanded manual extrusion widths, there is absolutely no change in the sliced print. The is incorrect behavior.

The only setting that actually nets a print change, is the nozzle width setting. This behavior is also incorrect, according to the tool tips, because if a manual extrusion width is specified, it should supercede the nozzle width setting. when set to 0.40mm default extrusion width, and 0.41mm nozzle width, I get 0.41mm extrusions.

CCS86, 2020-10-21:
Sorry, I was for some reason believing that the % expression referred to the default width, rather than the nozzle width.

CCS86, 2020-10-22:
It still seems odd to me that the user can't simply ask for a specific extrusion width and get it. Many of my prints end up with thin enough walls, that I have no infill, only perimeters. 

To test what is required, since I am doing this visually in SS, I went much wider than usual: 0.4 * 40 = 16mm

With perimeter overlap at the default 100%, I need to command extrusion widths of 0.431mm, to end up with 0.4mm extrusion spacing:

[image]

Wouldn't it make more sense to have the commanded extrusion width define the extruder stepover distance exactly, and have the perimeter overlap compensation, simply apply a background extrusion compensation? It seems like the end result would be the same, but with something far easier for the user to configure.

supermerill, 2020-10-22:
I've since added a help text below the extrusion width area (maybe in 2.2.53.3), to explain how to use these, and why width is not the same as spacing.

But there IS a problem, already acknowledged in my second comment.

CCS86, 2020-10-22:
I realize that you didn't create the "extrusion width" definition, or the fact that it is the primary, front-end setting.

Originally, I just misunderstood these definitions. Now, I question the merit of doing it this way. From the user point of view, it seems like the extrusion width value is arbitrary and should be internal. For any extrusions, bounded on both sides by other extrusions, the step over is really the most important thing to the user. For extrusions only bounded on one side, like internal/external perimeters, it seems like you would just apply ~1/2 the overlap compensation, so the exposed edge is still positioned correctly. Only in the rare case of a single wall segment, like a spiral vase, is the extrusion width ever observable (and I'd argue still not super important).

If you don't want to dig into that part of the code to overhaul, I understand. I have figured out how to "game" the settings, to get my desired result. But I think a lot of people would also be confused when it is so fiddly to get an exact wall thickness, made up of nice even widths.

supermerill, 2020-10-22:
That's not that i don't want to overhaul, but i have to ensure that it won't remove control to user that are accustomed to it and won't create problems.

CCS86, 2020-10-23:
I understand that need.

Since the stepover (line width) and extrusion width are interconnected (with a couple others involved like perimeter overlap, layer height, etc); a solution could be that every time you set one, the other is calculated.

Example, with overlap set to 100%, if you set stepover (line width) to 0.4mm, extrusion width would be set to 0.431mm.

supermerill, 2020-10-23:
not that simple, because it also depends on the external perimeter width. and it depends on the number of perimeters you want.

qwewer0, 2021-01-26:
Not sure if this helps, but I hope: https://github.com/supermerill/SuperSlicer/issues/682#issuecomment-729661236

qwewer0, 2021-01-26:
I hope my following question is connected to this issue, and that someone can enlighten me about this.

Why is the difference? (gap/no gap on the bottom left corner)
Every other setting is the same. (2.3.55) [Shape-Box 3mf](https://github.com/supermerill/SuperSlicer/files/5876759/Shape-Box.zip)
On both case, shouldn't it extrude the same amount and on the same path?
Why does the nozzle needs to be in the calculation if the extrusion width is specified separately from the nozzle?

2mm nozzle:
[image]

0.2mm nozzle:
[image]

CCS86, 2021-01-27:
> not that simple, because it also depends on the external perimeter width. and it depends on the number of perimeters you want.

I don't see that as breaking the request.

You just start at the outside and move inwards, right? Given the outside of the model, and the external perimeter width, you can plot the correct path. From there, you take the general perimeter width / line spacing, and plot the paths for the remaining perimeters. Using what is now labeled "overlap %", you would calculate the correct flow rate to fill voids using the user defined line spacing.

qwewer0, 2021-01-27:
> What were your line widths set to on these two prints? If the line widths were set the same, and you made a 10x change to the nozzle width, I can see why you would see some weirdness.

@CCS86 
The extrusion width was 0.4 for all, and the 10x nozzle was in order to exaggerate the effects, so that it can easily be seen.

I asked this, because it doesn't seem logical (or can't see the logic yet) to have the nozzle diameter be a part of the calculation to where a line ends.
The line width calculation works without the nozzle diameter (https://github.com/supermerill/SuperSlicer/issues/682#issuecomment-729661236) so why does the nozzle effects the line length?

supermerill, 2021-04-27:
Now you can set the spacing or width, as you wish.
Spacing is simpler and safer. Note that will modify the width if you change the layer height and it won't be correct if you're using variable height.

closed 2021-04-27 by supermerill

CCS86, 2021-04-27:
Awesome, can't wait to try it!

supermerill, 2021-05-27:
it's up

Tahx, 2022-01-26:
@supermerill What are the variables used to calculate the tooltip that recommend the object wall thickness?

My object is round and the outer wall is 2mm thick, but I need to set 0.395mm spacing to avoid having it putting a gap inbetween perimeters.

supermerill, 2022-01-26:
@Tahx 
`layer height`, `external perimeter width`, `perimeter width`

Tahx, 2022-01-26:
@supermerill Ahh so it's still using the perimeter width and not the spacing to do the calculation despite being able to set spacing instead now, it's all a bit confusing, I'm wondering whether I should just update the thickness of my object from now on...

I've never liked those gap fill that tends to appear inbetween perimeters, they often bring overextrusion issue, it sucks a bit to not be able to use simple round wall thickness.

I will assume that if my nozzle diameter is 0.4 and that the spacing is at 0.4, there will barely be any overlap?
So the bonding will suck UNLESS the squish on the bed is low enough, correct?
But then dimensional accuracy will suffer - the object will end up bigger, yeah or no?

Tahx, 2022-01-26:
@supermerill So to summary, the tooltip use the extrusion width to calculate the recommended wall thickness, but it also adds in the nozzle diameter minus the spacing.

In the case of an extrusion width of 0.4, the spacing is at 0.3571, leaving a value of 0.0429
For 5 perimeters (including external at 0.0215 value) that would be 0.236

which is what the tooltip recommends for 5 perimeters: 2.23mm

SO at the end, that means the actual extrusion width is 0.4429 and not 0.4mm

I'm not sure if I explain myself properly, but there's something that's not right here in the way it's shown.

supermerill, 2022-01-26:
spacings are a way to set the widths easily (for the current layer_height). Only the width is used internally.
If you set a spacing, the width is computed accordingly.

> I will assume that if my nozzle diameter is 0.4 and that the spacing is at 0.4, there will barely be any overlap?

the slicer will try to extrude extrusions smaller than your nozzle diameter.
The overlaps are set in the overlap section (and in the filament max overlap)

> actually there's an issue with the amount of perimeters given in the tooltip, the tooltip says to have a min wall thickness of 2.23mm for 3 perimeters, but that's actually 5 perimeters?

it has to go back.

CCS86, 2022-01-27:
> @supermerill Ahh so it's still using the perimeter width and not the spacing to do the calculation despite being able to set spacing instead now, it's all a bit confusing, I'm wondering whether I should just update the thickness of my object from now on...
> 
> I've never liked those thin walls that tends to appear inbetween perimeters, they often bring overextrusion issue, it sucks a bit to not be able to use simple round wall thickness.
> 
> I will assume that if my nozzle diameter is 0.4 and that the spacing is at 0.4, there will barely be any overlap? So the bonding will suck UNLESS the squish on the bed is low enough, correct? But then dimensional accuracy will suffer - the object will end up bigger, yeah or no?

No, you've got that wrong. Spacing of 0.4mm on a 0.4mm nozzle works beautifully, because the perimeter overlap setting is what controls width/overlap.

Post a picture of what result you are getting and what you want it to do instead.

supermerill, 2022-01-27:
note that in your last picture, the middle extrusion is maybe a gapfill that was transformed into a perimeter as it was detected as a loop.