supermerill/SuperSlicer #594: SuperSlicer quits when configuring Repetier.

thierryzoller, 2020-10-22:
### Version
Latest

SuperSlicer crashes when I try to configure Repetier (it quits without error message).
Are there debug logs somewhere?

NickRno77, 2020-10-22:
me too, it does this when I paste in the API Key.

supermerill, 2020-10-22:
no logs.
I didn't test the repetier, as i don't know how to use it. I trust the pull request.
Tell me how to reproduce, and i'll see how to fix it.

thierryzoller, 2020-10-22:
same then NickRno77

NickRno77, 2020-10-22:
> no logs.
> I didn't test the repetier, as i don't know how to use it. I trust the pull request.
> Tell me how to reproduce, and i'll see how to fix it.
> 
> I shouldn't have use repetier as the name for the release....
Similar to OctoPrint I think except there’s a server name option which I never get to anyway. I pick repetier from server list, enter IP address of the repetier server, enter the API key then crash....

Thanks

supermerill, 2020-10-22:
[image]

NickRno77, 2020-10-22:
> [image]
> 
> maybe i need a real repetier server to make it crash?

Yes I think you need valid API key, if I put 1234567 it doesn’t crash but obviously doesn’t connect.   Couldn’t  Boris Pruessmann help ?

Lukas1818, 2020-10-22:
I can reproduce this with my local repetier-server and the [live demo](https://www.repetier-server.com/live-demo/).
Until commit db99b091d97eb3a56a4d70de69c9595e3917162e  superslicer does **not** crash.
Only newer version since 3ec7dec14010f9bab2192622d84eefd2f86a3ddf crash.

Superslicer does not crash if you had configure repetier-server before commit 3ec7dec14010f9bab2192622d84eefd2f86a3ddf.

Lukas1818, 2020-10-22:
> Tell me how to reproduce, and i'll see how to fix it.

* Open the [live demo](https://www.repetier-server.com/live-demo/)
* press the "open repetier server live demo" button
* gear icon -> Global Settings -> Connectivity    (you can see the API-key  here)
* In Superslicer: change Host Typ, enter URL with port and the API-key from live the Demo

supermerill, 2020-10-23:
someone can test the last nightly? It's in the action menu.

NickRno77, 2020-10-23:
> Someone can test the lastest nightly? It's in the action menu.

that works, thanks

Lukas1818, 2020-10-23:
I test this too and it does work. Thanks

Please remember that I was not able to reproduce the crash, you had mentioned [here](https://github.com/supermerill/SuperSlicer/pull/569#discussion_r506736468).

supermerill, 2020-10-23:
that's a bug from an assert in wxwidget, it only happen in debug.
But i'm not confident enough to let it slip through, so I "fixed" it by removing a hacky ui-bugfix in this exact case.

closed 2020-11-25 by supermerill