supermerill/SuperSlicer #595: Fan Speed external Perimeters

thierryzoller, 2020-10-22:
### Version
Latest
Inconsistent Fanspeed results for external perimeter.

[image]
[image]

thierryzoller, 2020-10-22:
[image]

supermerill, 2020-10-22:
can i have your project file?

thierryzoller, 2020-10-22:
[Fanspeed.zip](https://github.com/supermerill/SuperSlicer/files/5423027/Fanspeed.zip)

supermerill, 2020-10-22:
that's because of the fan startup delay (in the printer tab). It's here to ensure the fan will be at 100% when you'll be at the internal perimeter again.

I created that to ensure the fan will be at full steam when doing bridges and overhangs, as my petg small bridges had a bad tiem without it. But now that we also have fan control for things like external and top, maybe I should restrict it to bridge/overhangs fan speedup only.

thierryzoller, 2020-10-22:
Interesting, probably it's best placed in the Filament FAN section rather than in the printer ?

If that's the case, why do you think do we have those individual "random" fan speed adjustemnts on these pillars?

[image]

thierryzoller, 2020-10-22:
The Opposite Pillar has this (They are mirrored 100%)

[image]

closed 2020-11-25 by supermerill