supermerill/SuperSlicer #596: Feature or Bug ?

knickohr, 2020-10-22:
### Version
2.2.54

### Operating system type + version
Win 7 64bit

### 3D printer brand / version + firmware version (if known)
PRUSA Bear (MK2,5s) with SKR Turbo 1.4 and Marlin 2.0.7.x

### Behavior
Just a short question :

Is this a feature or a bug ?  I notice that the fan will speed up full power while heating the printer and doing the mesh bed leveling. This in, in my opinion, not the best thing because it harm the mesh bed leveling procedure.

; generated by SuperSlicer 2.2.54 on 2020-10-22 at 13:05:38 UTC

; 

; external perimeters extrusion width = 0.42mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.48mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.45mm
; first layer extrusion width = 0.45mm

M73 P0 R105
M107
===>>> M106 S255 ; speed up fan for testing
M150 R255 ; Red Light
M83  ; extruder relative mode
M104 S170 ; set extruder temp
....

How can I disable it ?

knickohr, 2020-10-22:
Upps, sorry, ist my own code, but this could be really a bug :

(output gcode)
; generated by SuperSlicer 2.2.54 on 2020-10-22 at 13:22:27 UTC

; 

; external perimeters extrusion width = 0.42mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.48mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.45mm
; first layer extrusion width = 0.45mm

M73 P0 R105
M107
===>  M106 S255 ; speed up fan for testing
|   M150 R255 ; Red Light
|   M83  ; extruder relative mode
|   M104 S170 ; set extruder temp
|   M140 S85 ; set bed temp
|   M109 S170 ; wait for extruder temp
|   M190 S85 ; wait for bed temp
|   G28 ; home all without mesh bed level
|   G29 S4000 ; ABL
|   M104 S230 ; set extruder temp
===<<<  this line moved to top   M106 S255 ; speed up fan for testing
M300 S4000 P1000 ; beep one second to clean nozzle
G1 X-10.0 Y-6.0 F2000.0 ; go outside print area
M106 S0 ; slow down fan
M109 S230 ; wait for extruder temp
M150 W255 ; White Light
G92 E0.0
....

original start gcode in slicer :

M150 R255 ; Red Light
M83  ; extruder relative mode
M104 S170 ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M109 S170 ; wait for extruder temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
G28 ; home all without mesh bed level
G29 S4000 ; ABL
M104 S[first_layer_temperature] ; set extruder temp
>>> M106 S255 ; speed up fan for testing
M300 S4000 P1000 ; beep one second to clean nozzle
G1 X-10.0 Y-6.0 F2000.0 ; go outside print area
M106 S0 ; slow down fan
M109 S[first_layer_temperature] ; wait for extruder temp
M150 W255 ; White Light
G92 E0.0

Thy is this line with M106 move yourself up ?

knickohr, 2020-10-22:
Ahhh, found it ;)

It's the new "Fan startup delay".

So, my next Feature Request :  Disable it in the start gcode !

supermerill, 2020-10-22:
maybe also on all custom gcode sections

knickohr, 2020-10-22:
Yepp, thats make sense ;)

closed 2020-11-25 by supermerill

knickohr, 2021-03-04:
The issue is back !  :o

Version 2.3.55.5