supermerill/SuperSlicer #597: monotonous crash the slicer

supermerill, 2020-10-23:
### Version
2.2.54.1

### Operating system type + version
win

### Behavior
when slicing darth-vader-bust-body-by-eastman.3mf with monotonous (filled) on bot & top, it crash.

skellatore, 2020-10-24:
Doesn't seem to do it if you just use Monotonous just on the bottom layer

Matts-Hub, 2020-10-28:
Also, setting infill to 100% with monotonous pattern selected causes a crash for me with any model

supermerill, 2020-10-28:
If you can link the projects where it crahsh, currently i only have on and the object is very big. It's super slow to debug.

Matts-Hub, 2020-10-28:
Sure thing, here it is. Just bump infill % up from 99 to 100 and it will crash
[x_end_idler.zip](https://github.com/supermerill/SuperSlicer/files/5450947/x_end_idler.zip)

supermerill, 2020-10-28:
Seams like prusa has also discover and correct it.
Their patch for yesterday works well.

closed 2020-11-25 by supermerill