supermerill/SuperSlicer #598: Auto center parts causes calibration files to be messed up

pheonix3330, 2020-10-23:
### Version
2.2.54

### Operating system type + version
Windows 10 64 bit

### Behavior
When generating calibration prints, if "Auto center parts" is enabled in preferences, the parts will not fit on the plate. It seems like the slicer tries to auto center parts in between instances of the calibration items being generated, causing everything to shift out of place.

To reproduce, enable "auto center parts" in preferences and then generate a bed level calibration print. The expected result is one object in the center and one object at each corner. The actual result is several objects outside of the build volume.

I believe "auto center parts" should not apply to calibration files generated by superslicer and that would solve the issue.

supermerill, 2020-10-23:
good catch, I never use this so I didn't think of it as a possible problem.

thierryzoller, 2020-10-28:
Finally! This also fixes my 2 previously reported issues.

closed 2020-11-25 by supermerill