supermerill/SuperSlicer #599: Feature request: dont drop z before x-y is in pos at part

Coffee0297, 2020-10-23:
### Version
All
### Operating system type + version
win10

### Behavior
I sit possible we can make so Z don't drop before the printhead is at the starting line? Atm its like this:
G92 E0
M107
G1 Z0.120000 F7800.000
G1 E-4.00000 F2400.00000
G1 X99.248 Y99.462 F7800.000

ideal movement wuld be:
G92 E0
M107
G1 E-4.00000 F2400.00000
G1 X99.248 Y99.462 F7800.000
G1 Z0.120000 F7800.000

Its not critical but a nice thing.

there is also the labeling, the movement from 1 part to another is before the label, that shuld idealy be after the lablel so if i cancel that part it should not go to the canceled part and back as it do now. Again not critical but a QOL.

supermerill, 2020-10-23:
Difficult to modify (without creating bugs).

foreachthing, 2020-10-23:
The main reason I made this here: https://github.com/foreachthing/Slic3rPostProcessing

These are the changed lines:
<pre>
M117 [.                 ]
G1 E-6.00000 F3000.00000
G92 E0
G1 X97.190 Y96.130 Z0.200 F6000.000 ; move to first brim point
G1 E6.00000 F3000.00000
G1 F1800.000
;segType:Skirt
;TYPE:SKIRT
G1 X98.983 Y94.671 E6.03863
</pre>

The printhead moves from 
`G1 X1 Y5.1 F15000	; move X/Y to start position` (my startpoint) to 
`G1 X97.190 Y96.130 Z0.200 F6000.000 ; move to first brim point` in one straight line.

closed 2021-11-19 by supermerill