supermerill/SuperSlicer #600: Feature Request

digitalentropy, 2020-10-23:
### Version
N/A

### Operating system type + version
N/A

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Sometimes it would be nice to know if changing to a larger nozzle size would meaningfully lose any detail in a given part. I don't know how difficult it would be to handle this in slicing, but some sort of analysis tool to recommend the maximum "ideal" nozzle size for a given project could be helpful.

Makhaira, 2020-10-27:
I think you're going to struggle with how subjective this is - arguably every increase in nozzle size results in loss of detail because the corners become rounder. I'm not sure how you'd specify a threshold for this in a slicer.

digitalentropy, 2020-10-27:
> 
> 
> I think you're going to struggle with how subjective this is - arguably every increase in nozzle size results in loss of detail because the corners become rounder. I'm not sure how you'd specify a threshold for this in a slicer.

Thinking about it more, you're absolutely right. There's no way for the slicer to discriminate between "features that matter" and "features that don't".

closed 2020-10-27 by digitalentropy