supermerill/SuperSlicer #601: Feature Request: transparency/opacity option

madsi1m, 2020-10-24:
@supermerill here is a super low priority feature request that could be super cool for super slicer.

Add a transparency/opacity option or slider to the preview page so I can see through the solid layers without needing to adjust the layer sliders. Would be a good way to see inner voids and infill overall.

PS: Keep up the good work!

neophyl, 2020-10-24:
Second this, in fact I made the same feature request over for PS awhile ago.  I think I called it xray mode.  
Overlapping parts could be spotted that way too as it could make it more opaque and I have still seen new users load a part in twice without realizing it as they are both in the centre and the resulting prints are not pretty.

madsi1m, 2020-10-24:
paste in the PS link to the issue might be helpful

supermerill, 2020-10-24:
I've not touched to the shader/opengl yet. low low priority for me.

madsi1m, 2020-10-24:
Could be a fun way into the shader if/when you do

supermerill, 2022-03-23:
#2369