supermerill/SuperSlicer #602: Feature Request: Move Z-Offset to Filament Section

sverzijl, 2020-10-24:
Version 2.2.53

Windows 10

Prusa MK3S

PETG and Nylon benefit from an increased z-offset. Currently it is located in 'Printer Settings', however it would be beneficial to have an override setting in Filaments so that it can be made dependent on the filament.

supermerill, 2020-10-24:
???
how can a filament change the position of the toolhead?

sverzijl, 2020-10-24:
Hi,

As an example, PETG likes to stick to the nozzle. If you optimise the bed
height using PLA and then print PETG you can see a build up of PETG
occuring on the nozzle. This leads to globs of plastic falling on the model
and excessive stringing. You can resolve this issue by using a z-offset of
about 0.2 and then correcting for issues on the first bed by increasing the
extrusion multiplier on the first layer.

Thanks!

1N4148, 2020-10-24:
PETG may require a greater offset for the first level to prevent it to be to sticky to the heatbed. 
I've solved this with the filament gcode to use an additional offset of 0.1mm for PETG. 
But, the Z-Offset in the printer is still needed for fine tuning the general nozzle offset.
If a filament offset is added it should be in addition to the printer's Z-Offset, not instead of it.

1N4148, 2020-10-24:
@sverzijl In your case I assume your filament extrusion multiplier isn't correct. With the correct multiplier there shouldn't be any excessive material.
I'm running PLA with 0.9 multiplier and PETG with 0.95 multiplier and getting perfect first layers with the same nozzle height. No excessive material which is collected by the nozzle.

But for PETG it's better to have a small underextrusion at the first layer to prevent it to be to sticky to the bed.

sverzijl, 2020-10-24:
@1N4148 Extrusion multiplier is correct (vase mode calibration cube and micrometer).  The idea of having a slightly higher Z-offset to prevent build up on nozzles comes up quite often. An example: https://rigid.ink/blogs/news/175700615-petg-filament-heres-what-you-need-to-know 

The guidance was basically what allowed me to print PETG exclusively and move away from PETG.

The reason why I then increase the extrusion multiplier is to compensate for under extrusion on the first layer. I use a textured PEI sheet so I have no issues with it being too sticky to the bed.

I shouldn't have said move - I think it just needs to have an override.

supermerill, 2020-10-24:
I don't think that's the right override.
if a first layer extrusion multiplier (or even better, width) can do the job, it's imo best as it won't have as much as side problems/bugs as a z-offset.

sverzijl, 2020-10-24:
Some more:
https://www.3dprintingsolutions.com.au/User-Guides/how-to-3d-print-petg-filament#Bed%20Leveling%20and%20Nozzle%20height
https://standardprintco.com/read/2020/11/15/petg-filament-printing-guide-how-to-succeed-printing-petg-and-troubleshooting-common-problems-step-by-step
https://forum.prusaprinters.org/forum/original-prusa-i3-mk2-s-how-do-i-print-this-printing-help/setting-live-z-with-petg/
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/another-petg-question/
https://3dprinting.stackexchange.com/questions/8059/petg-nozzle-clearance-and-extrusion-multiplier

sverzijl, 2020-10-24:
@supermerill The extrusion multiplier only compensates for the extra z-offset. I've confused things by mentioning it. All I am talking about is being able to add additional z-offset based on the filament.

supermerill, 2020-10-24:
in my prusa printer (mk2), i had to adjust the z-offset for petg because the difference in temperature modify the response of the inductive sensor.
I fixed that by doing the mesh bed levelling before heating the the bed.

1N4148, 2020-10-24:
The first layer extrusion multiplier isn't linked with the filament so I would rather use the filament specific gcode to add an additional offset.

TD5023, 2020-10-24:
You can easily achieve an additional, filament-specific offset with a G92 command.

G0 Z{first_layer_height} ;Move to default first layer height
G92 Z{first_layer_height - OFFSET} ;Input offset value (replace OFFSET with an actual number)
G0 Z{first_layer_height} ;Move to adjusted first layer height

I have to have a few more numbers thrown in because of complications with M206, but as long as you don't use that and it's set to zero, you won't have a problem.

1N4148, 2020-10-24:
For Klipper users: SET_GCODE_OFFSET Z_ADJUST=0.05

digitalentropy, 2020-10-27:
I agree, I think it would be nice to add a z-offset override checkbox to the filament-specific overrides tab.

Makhaira, 2020-10-27:
I've done this on both Marlin and Klipper, and considering how simple it is to implement in the custom gcode I'm not sure of the benefit of loading this up as another feature to maintain in the software?

Liger0, 2020-10-29:
I never understood why people keep suggesting to use a z-offset or x-y width compensation to circumvent other issues. 
For petg you just need to increase the first layer height.. A z offset will only make for a wrong Z measure, and gaps if the flow is correct.
This said, it would make sense for ABL probes, like capacitive ones, which can detect a bit differently depending on the heat of the bed, making the z offset slightly different between 50°C and 110°C for ABS (I think prusa has a pid calibration for their sensors because of this reason, but I'm not sure)

TD5023, 2020-10-29:
Because increasing the first layer height also adjusts the flow to compensate, which for many surfaces, makes it sick just as hard. Increasing the gap without increasing the flow (i.e. Z-offset) is the way to actually make it stick less, and for most practical applications, the difference in height of the object is negligible.

sverzijl, 2020-10-29:
@Liger0 
That's another approach - though does that maintain the increased distance between the nozzle and the print for subsequent layers?

I guess my only comment to that is that first layer height isn't in the filament section either even if more appropriate.

Not on topic but that is what I find unsettling about Slic3r/SuperSlicer/Prusa Slicer profiles - for many filaments you need to have filament specific settings for Print Settings and Printer Settings. For example bridging parameters, which very much are filament specific, are in 'print settings'

Liger0, 2020-10-29:
> Because increasing the first layer height also adjusts the flow to compensate, which for many surfaces, makes it sick just as hard. Increasing the gap without increasing the flow (i.e. Z-offset) is [...]

Then a first layer flow would make more sense, at the cost of small gaps.

supermerill, 2020-10-29:
> Then a first layer flow 

that already exist.

TD5023, 2020-10-29:
@sverzijl I would love it if we could pull in parameters for overrides section just like you can for the individual objects on the plater screen. That would solve quite a few issues on the user side.

@Liger0 that should have a similar effect (in theory, at least). I've tried something similar by making the width thinner, but it didn't help enough. I think the extrusion overlap still made it stick too strongly, but I don't know. I just know I got the best and most consistent results by adjusting the Z a tiny bit.

khumarahn, 2020-10-29:
It is easy to set a gcode offset in printer start gcode and make in conditional on the type of filament. E.g. for klipper:
```
{if filament_type[0] = "PETG"}
SET_GCODE_OFFSET Z=0.1
{endif}
```

rngSplat, 2020-11-23:
I would also appreciate filament specific Z-offset, since I (and others) modify it for PETG. Certainly not everyone does it that way, but enough people do that I think it warrants inclusion.

Though I did just switch to Klipper, so there are some handy workarounds in this thread.

GoodyGizmos, 2021-03-06:
@supermerill I would also like to see this implemented in the filament settings.

@khumarahn I find different PLAs and even different spools require slight adjustments to the Z offset, so I wouldn't want it implemented my filament type alone.

khumarahn, 2021-03-06:
> @khumarahn I find different PLAs and even different spools require slight adjustments to the Z offset, so I wouldn't want it implemented my filament type alone.

Then you can `SET_GCODE_OFFSET` in the filament start gcode, it is a very robust and very existing solution.

GoodyGizmos, 2021-03-06:
@khumarahn Thanks. Just to test it, I added SET_GCODE_OFFSET Z9.9 to the filament Start G-code, resliced, and sent it. Of course, I expected it to print in mid air. After the wipe, it rose to what appeared to b 9.9, then re-homed and started printing without an offset.

martinbudden, 2021-06-21:
For non-klipper users, you can add the following to your Start G-code to set the Z offfset to 0.05 (say) for PETG
```
{if filament_type[0]=="PETG"}
G1 Z0.05 ; move to Z offset for PETG
G92 Z0 ; set Z=0 at Z offset for PETG
{endif}
```

supermerill, 2021-06-21:
```
{if first_layer_bed_temperature[initial_extruder] > 80}
G1 Z0.05 ; move to Z offset for ABS
G92 Z0 ; set Z=0 at Z offset for ABS
{elsif first_layer_bed_temperature[initial_extruder] > 60}
G1 Z0.025 ; move to Z offset for PETG
G92 Z0 ; set Z=0 at Z offset for PETG
{endif}
```

```
{if first_layer_bed_temperature[initial_extruder] > 80}
SET_GCODE_OFFSET Z=0.05
{elsif first_layer_bed_temperature[initial_extruder] > 60}
SET_GCODE_OFFSET Z=0.025
{endif}
```

gpread, 2021-11-29:
This would seem like a useful addition to me.

mordant80, 2021-12-25:
This feature would be very helpful since different filaments require different bed temperatures.  I've found the different bed temps do require different offsets.  I dont know how much of this is from thermal expansion, but i'm sure it plays a roll depending on the bed surface materials and thickness.  Would at least like to see it as an expert option in Filament Overrides.