supermerill/SuperSlicer #603: Slicing errors: part of model ignored

khumarahn, 2020-10-25:
I could reproduce this with the a few most recent appimages, on default settings for random printers, such as prusa mini.

I am trying to slice a model
[image]

And I get
[image]

Inspecting the model in blender I don't see anything wrong. Cura slices it fine.
[CatFace_Relief.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5434578/CatFace_Relief.3mf.zip)

neophyl, 2020-10-25:
Not sure how you are checking in blender but when I check it in blender it reports - 193 intersecting faces, 112 zero faces,  9 zero edges,  29 non flat faces and 2 thin faces.  Also Super Slicer is trying to repair it and fixes 48 errors so there is obviously errors in the model.

Whats really strange though is I used blender to make manifold (one button auto click fix attempt) and loaded it into Super Slicer 2.2.54.1 along with the original and both of the sliced fine, the original from your saved 3mf file and the one I modified.  So I'm not sure why you are having trouble.  
Just in case I'll attach my fixed version which does load into slicer without showing errors (although Blender still shows some anyway).
[CatFace_Relief_neo.zip](https://github.com/supermerill/SuperSlicer/files/5434846/CatFace_Relief_neo.zip)

supermerill, 2020-10-25:
it can also be repaired via the File -> repair stl

khumarahn, 2020-10-25:
@neophyl @supermerill 
sorry for the wrong alarm! How to you check that this model is invalid in blender? I look for non-manifold elements but do not find any.

khumarahn, 2020-10-25:
In superslicer I do not see that it is not a manifold, and repair manifold does not help much. Probably this file is indeed too invalid.

neophyl, 2020-10-25:
Super Slicer has a yellow triangle check mark next to it to warn you that it has fixed some errors.  Unfortunately it can’t fix everything but it does often fix enough to slice. There’s on online service called Netfabb that is also very good at fixing issues but even it can’t fix everything. 

In blender there’s one of the built in add ins you can turn on called 3D print tool. It has tools for checking issues as well as a make manifold button. That doesn’t always work though so manual intervention is called for in those cases. In this case the make manifold button fixes it enough so that when imported into Slicer it doesn’t show any errors and slices fine.

khumarahn, 2020-10-25:
@neophyl Thanks a lot, I am going to explore the manifold validation things. I am closing this issue because it is not a problem with superslicer.

closed 2020-10-25 by khumarahn