supermerill/SuperSlicer #605: Bug?: "solid Infill" Allways on with 2 Perimeters

Zulaner, 2020-10-26:
### Version
2.2.54

### Behavior
Even disabling all options for solid fill, always appearing.
[image]
[3DBenchy.zip.amf.zip](https://github.com/supermerill/SuperSlicer/files/5438385/3DBenchy.zip.amf.zip)

neophyl, 2020-10-26:
That's because its not normal solid infill.  Its a known issue in Prusa Slicer and is related to vertical shell thickness on sloping walls but it happens regardless of if that option is checked or not too.  So yes technically a bug but exists in Prusa Slicer currently and historically.

supermerill, 2020-10-26:
duplicate of https://github.com/supermerill/SuperSlicer/issues/253

closed 2020-10-27 by Zulaner