supermerill/SuperSlicer #608: [feature request] Fan blip for 5015 fans

Liger0, 2020-10-28:
###3D printer brand / version + firmware version (if known)
Cr10S, marlin 2.0.x, 5015 fans

### Behavior
It would be very useful to have a fan blip feature like s3d does.
This would help with 5015 fans and materials like petg, where you usually print with fans off, but turn them on to low values (like 30%) for bridging and external perimeters.
So the fan, from an off state, would go to 100% for a fraction of second, just the minimum time necessary to reach such speed, and then go down to the desired speed.

supermerill, 2020-10-28:
This is often done by the firmware.
kick_start_time in klipper
FAN_KICKSTART_TIME in marlin/repetier
reprap surely have something similar.

or not?

Liger0, 2020-10-28:
Hmm... yes, it seems Marlin has it. 
I will have to test it, although it may still be useful for people that still have stock firmwares and even don't want to flash a bootloader if the machine doesn't come with one.
Marlin says "Does not work with software PWM for fan on Sanguinololu)". I am not sure what it is, but I guess modern motherboards are not related to that, since 5015 fans also need the software PWM.

supermerill, 2020-10-28:
i'll see if it's not too hard to add it to the current fan_delay feature.

madsi1m, 2020-10-31:
Would be good if we had a tickbox to spin the fan at 100% for the duration of fan_delay. That said I current use fan_delay + firmware kick start and all is well

supermerill, 2022-03-23:
there is a kickstart now.

closed 2022-04-01 by supermerill