supermerill/SuperSlicer #610: Infill/perimeters overlap computes overlap thru whole shell

mocksoul, 2020-10-29:
### Version
Nightly build 28 Oct 2020

### Operating system type + version
Windows 10 64

### 3D printer brand / version + firmware version (if known)
Klipper / voron

### Behavior
Infill/perimeter overlap seems to overlap thru whole shell in superslicer. Thus, default setting of 25% means 0.5mm in 2mm shell consisting of 5x0.4mm perimeteers.

100% overlap, SuperSlicer (nightly, 28 Oct 2020):

[image]

In PrusaSlicer this is properly computed against inner perimeter only, so 25% overlap means just 0.1mm overlap with 0.4mm perimeter line width.

100% overlap, PrusaSlicer (2.3.0a2)

[image]

This leads to huge overextrusions of perimeters, visual artefacts even in external layer with bigger infill/perims overlap [values.]

mocksoul, 2020-10-29:
After few more testing it looks like in superslicer infill/perims overlap multiplied by some value. Try 2mm overlap with 1mm shell to see weird results in preview :-D

supermerill, 2020-10-29:
can i have your project file?

mocksoul, 2020-10-29:
sure
but what is "project file"? Stl? Or what? ))

STL: https://github.com/VoronDesign/VoronUsers/blob/master/printer_mods/Slidr/PanzerChain2.4/stl/z_anchor_frame_vertical_solid_300_x1.stl

Also maybe this is only gcode-preview isue, bc it didnt not fix overextrusion after I set 0% overlap =(

Problem is the grid infill pattern. After switching it off, or using gyroid (which is almost like "switch off" in this stl :)) -- all issues gone. Even in plater window no noticable issues with perims/infill overlap. But with grid in fill it is still weird :-D

PrusaSlicer has same behaviour, except plater window -- no strange overlapping with grid infill there.

neophyl, 2020-10-29:
Project file is the saved .3mf file that is produced when you select file>save project. It not only contains the model file but all the settings used to slice, so print, filament and printer profile settings as well as any modifiers and the model position etc on the plate.

mocksoul, 2020-10-29:
Here it is. I set grid infill again and 100% overlap -- same settings above screenshots were made.

[z_anchor_frame_vertical_solid_300_x1.zip](https://github.com/supermerill/SuperSlicer/files/5460343/z_anchor_frame_vertical_solid_300_x1.zip)

mocksoul, 2020-12-01:
I cant reproduce this with latest mater either, closing

closed 2020-12-01 by mocksoul