supermerill/SuperSlicer #611: Maybe create Discord server?

cab404, 2020-10-29:
It would be nice to have a community hub around the best fork of the best slicer)

wildjokerss, 2020-10-29:
I second this. Actually in the past I've been looking for it as I thought it already existed :D

supermerill, 2020-10-29:
https://discord.gg/QaJj92

I may need some people willing to do the moderation stuff.

cab404, 2020-10-29:
> https://discord.gg/QaJj92
> 
> I may need some people willing to do the moderation stuff.

I've pinged you about that. I'm cab.

homewerken, 2020-10-30:
Is the discord Server still Open ?

supermerill, 2020-10-30:
it can close?

closed 2020-11-25 by supermerill

supermerill, 2021-08-29:
as some bots have discovered the previous link, here a new one.
It's only the key, you have to add the 'discord.gg' before it:

`4TVUG4VRzP`

D-Dub24, 2022-04-09:
The link appears to be expired again.

DatenThielt, 2022-04-13:
> https://discord.gg/ygBBdRRwJY
> 
> I may need some people willing to do the moderation stuff.

Read the messages above, there is a valid link that is just missing the discord.gg part, due to bots

cab404, 2022-04-13:
updated the top comment, just so it's more obvious

D-Dub24, 2022-04-13:
> > https://discord.gg/ygBBdRRwJY
> > I may need some people willing to do the moderation stuff.
> 
> Read the messages above, there is a valid link that is just missing the discord.gg part, due to bots

I figured it out, but when I asked on the Voron discord, someone told me, regardless of if I figured it out or not, the next person that comes along is going to probably do the same thing. So they told me to post the issue so it gets fixed so others do not make the same mistake.

Thank you for making it more obvious for us illiterate people.

psi-4ward, 2023-02-15:
So do we have an discord? Seems all invites are expired

cab404, 2023-02-16:
> So do we have an discord? Seems all invites are expired

I think the one in the top message should still be valid :\

psi-4ward, 2023-02-16:
Es it's working. Thx.
Why not simply update the badge-button in README?

vkhodygo, 2023-05-06:
@cab404 it's dead again

Creo2005, 2023-07-16:
> @cab404 it's dead again

https://discord.gg/4TVUG4VRzP

Creo2005, 2023-07-20:
> 

NO! 
just checked

vkhodygo, 2023-07-22:
@Creo2005 
Bloody dicsord, it needs the app to be installed to handle the invite correctly.

cab404, 2023-07-22:
> > 
> 
> NO! just checked

Open in a private browser link and report, please

Creo2005, 2023-07-22:
> https://discord.gg/4TVUG4VRzP

[image]
sorry - screen in russian.