supermerill/SuperSlicer #612: Artifacts when XY-compensation and Holes compensation are enabled.

Slatkey, 2020-10-29:
### Version
2.2.54.2

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Zav3D. Klipper firmware 0.9.1, mainsail interface.

### Behavior

[image]
[image]

When the XY and / or hole size compensation functions are activated, artifacts appear on the model, which are then transferred to the printed model.

#### Project File (.3MF) where problem occurs

[side_fan_support_x2.zip](https://github.com/supermerill/SuperSlicer/files/5458108/side_fan_support_x2.zip)

supermerill, 2020-10-29:
your zip is corrupted

supermerill, 2020-10-29:
that's because of the hole.
for these layers, there is no more "internal" space.

1) Holes compensation isn't useful for this model, so you should deactivate it : it only apply to areas and layers where these big hexagon are.
2) If you need the xy compensation, set the same value for both outer & inner.

Slatkey, 2020-10-29:
I was more worried about XY compensation, your advice to set the same value for internal and external did fix the problem! Thanks for the answer

supermerill, 2020-10-29:
i'll update the tooltips to makes it more obvious.

closed 2021-06-14 by supermerill