supermerill/SuperSlicer #613: can't export Configuration

SidewalkPsycho, 2020-10-29:
### Version
V2.2.53

### Operating system type + version
Win 10 64 bit

### Behavior
Hi there, i can't export my configuration. I get the error message : value out of range: filament_shrink.
but it's at 100% as you can see in the attached screenshot.
https://prnt.sc/v9h57m

supermerill, 2020-10-29:
seems like a duplicate of #528

Did it also happens with the 2.2.53.3 ?

timoleary, 2020-10-30:
> seems like a duplicate of #528
> 
> Did it also happens with the 2.2.53.3 ?

Can confirm the problem exists with SuperSlicer_2.2.53.3_win64_20201005

StefanoPerinetti, 2020-10-30:
Yes I can confirm that the problem exist also in latest Mac Version SuperSlicer_2.2.53.3_macos_20201005.dmg

StefanoPerinetti, 2020-10-30:
[image]

[image]

[image]

StefanoPerinetti, 2020-10-30:
I download the precompiled file from github releases but I don't know if it contain the latest version
[image]

supermerill, 2020-11-02:
ok, it was fixed the day after the 2.2.53.3 release... time for a 2.2.53.4

StefanoPerinetti, 2020-11-03:
Ok thank you!

closed 2020-11-25 by supermerill