supermerill/SuperSlicer #614: Accelaration control "Default" based on printer settings

wildjokerss, 2020-10-29:
### Version
2.2.54

### Behavior
I often mess around with my acceleration and so on. The Acceleration control on the "print settings" is grayed out unless you set a default in there. Would be good I'd say to have a checkbox to "inherit" the acceleration from the printer settings, so whenever I mess the printer settings the print settings remain the same. 
Why am I asking this. All the other settings are grayed out without the Default one, however, I want to change only bridges to some way slower acceleration. The problem with setting a default there is that I might end up changing the Acceleration on printer settings and forget that I still have a different default.

[image]

This is indeed a feature request :D Let me know if I can clarify this any better :)

supermerill, 2020-10-29:
i can set it as a mm/s² or % ?

wildjokerss, 2020-10-29:
well yeah, that would be quite dope too! setting % based on printer settings acceleration

supermerill, 2020-10-29:
I'm using the max of the x/y max, or the min of the max ?
min should be safer.

wildjokerss, 2020-10-29:
[image]
I would suggest the printing and not the maximum @supermerill, I believe it would be a better, safer idea. Maximum could be there for travel purposes for example

wildjokerss, 2020-10-29:
So, to clarify, the Default acceleration on the print profile, expressed as a percentage, would grab data from the "Maximum acceleration when extruding"

closed 2020-11-25 by supermerill