supermerill/SuperSlicer #616: Perimeter bridge threshold incorrect when updateing from 2.2.53.2 to 2.2.54.2

yschroeder, 2020-10-30:
### Version
SuperSlicer_2.2.54.2_win64_20201028

### Operating system type + version
Windows 10

### Behavior
`'As bridge' threshold` is converted to `bridge flow` when updating from 2.2.53.2 to 2.2.54.2. I think it should be transitioned over to `bridge speed and fan` as that resembles what `'As bridge' threshold` did in 2.2.53.2.

Settings in 2.2.53.2:
[image]

Settings in 2.2.54.2:
[image]

supermerill, 2020-10-30:
It should be transitioned to both, as it's a split.

robthide37, 2020-11-01:
im lost isn't that what is done is this a problem or that label was before the last commit