supermerill/SuperSlicer #617: side lines on prints where geometry changes

JustinCaseOfWolf, 2020-10-30:
### Version
2.2.54

### Operating system type + version
Windows 10 64 bits

### 3D printer brand / version + firmware version (if known)
Voron 2.4

### Behavior
Print shows lines where geometry changes however the same print with kisslicer doesnt present the lines.
[z_tensioner_bracket_a_x2 kisslicer.zip](https://github.com/supermerill/SuperSlicer/files/5465278/z_tensioner_bracket_a_x2.kisslicer.zip)
[z_tensioner_bracket_a_x2 superslicer.zip](https://github.com/supermerill/SuperSlicer/files/5465280/z_tensioner_bracket_a_x2.superslicer.zip)
[image]
[image]

#### Project File (.3MF) where problem occurs

[z_tensioner_bracket_a_x2.zip](https://github.com/supermerill/SuperSlicer/files/5465262/z_tensioner_bracket_a_x2.zip)

yschroeder, 2020-10-30:
The lines coincide with the speed of the part cooling fan:
[image]

Kisslicer  probably has a different cooling strategy/fan settings. 

Basically, due to printing a single small part you run into minimum layer time and slicers starts to reduce speed (in infills first) and cranks up the fan speed. That might cool 

Either reduce the maximum fan speed that is allowed or print multiple parts at once (to increase overall layer time). Also, ABS should problably printed without any fan at all to avoid warping.

JustinCaseOfWolf, 2020-10-30:
the print in the image above is with fan 0, let me see if i can find the gcode for that one. the print with the fan on is this image:

JustinCaseOfWolf, 2020-10-30:
[image]

JustinCaseOfWolf, 2020-10-30:
[image]

yschroeder, 2020-10-30:
I downloaded your very last 3mf file `z_tensioner_bracket_a_x2.zip` and made the screenshot from that file. It definitely uses the part cooling fan.

Here is a similar screenshot of the part cooling fan operation from your SuperSlicer gcode:
[image]

And here is the part cooling fan usage from your Kisslicer gcode:
[image]

Still, the layer lines visible in your very first picture coincide with the part cooling fan operation in your gcode. SuperSlicer has the fan turned off most of the time (dark blue areas) and occasionally turns it on. Thus creating the artifacts.

However, I am not sure if SuperSlicer should turn the part cooling fan on at all at those locations. It uses the part cooling fan on some perimeters where I don't see a reason to do so.

As a workaround, you should probably set the part cooling up in the same way like you have in kisslicer. It seems to run at 30% all the time. However, I still think this is a bad idea for ABS in the first place.

Maybe SuperSlicer should not crank up the part cooling fan for outter perimeters to maintain the same surface finish?

JustinCaseOfWolf, 2020-10-30:
i feel i've managed to confuse you and for that i do apologise.

the first super slicer  print in the first picture above is made with this cooling settings:

[image]

second picture is kisslicer

third picture above is superslicer with the settings in the project.

i hope that makes sense - i;ve noticed the first print was missing cooling and made the changes for the second attempt and this is the point you asked me for the project. hope this makes it a bit clearer

yschroeder, 2020-10-30:
If the first print was printed with the gcode in the first post, it clearly had the part cooling fan running, according to the gcode. See my first screenshot in my second post.

Also the part cooling in the project seems to match the gcode in the first post. 

I am still confused. Can you try to explain the whole thing again from scratch?

JustinCaseOfWolf, 2020-10-30:
what i did was:
1) print with prusa slicer - terrible line
2) print with kisslicer - way better looking (as per picture up)
3) someone suggested to try superslicer which i did - and i printed without the cooling as per settings attached. this resulted in the first picture in this thread.
4) while it was printing i noticed the cooling was not on so i attempted to turn it on. slice, save (sadly overwrite the gcode)
5) been asked to create this issue report which i did but i stupidly included the original print with the modified gcode & project.
6) the new gcode has finished printing which is the new picture i've attached later in the thread.

hope this makes it a bit clearer

JustinCaseOfWolf, 2020-10-30:
[z_tensioner_bracket_a_x2 original.zip](https://github.com/supermerill/SuperSlicer/files/5468084/z_tensioner_bracket_a_x2.original.zip)

this is the gcode for the first superslicer picture

digitalentropy, 2020-10-31:
For what it's worth I have noticed the same issue printing PETG, ABS, and PC on a Haribo 3030 and SuperSlicer. I print ABS and PC with zero cooling fan for the whole job, and I still get similar lines where geometry changes.

1N4148, 2020-10-31:
Looks like the flow isn't matching the various speeds used.

JustinCaseOfWolf, 2020-11-02:
is there anything i can try to help with this issue?

supermerill, 2020-11-02:
I don't really know why it does that.
I'm going to do 2 things to try to resolve it:
* remove thick gapfill
* an option to change the periemter order : second -> first -> the others

supermerill, 2020-11-02:
@JustinCaseOfWolf can you try to reprint it with superslicer but without infill ? or with grid?
If it improves it, can you try then with gyroid but without the connection for the infill ?

JustinCaseOfWolf, 2020-11-03:
not sure what it means without the connection for the infill.

kisslicer has the option to specify 4,5 perimeters which means alternate 4 with 5 perimeters. it is supposed to make parts much stronger...

JustinCaseOfWolf, 2020-11-03:
i think there needs to be a bit of cooling but for some strange reason i cant get it to work. most of the print cooling seems to be 0%

JustinCaseOfWolf, 2020-11-03:
[image]
0% infill

JustinCaseOfWolf, 2020-11-03:
[image]
grid infill, 42%

supermerill, 2020-11-03:
seems like the 0% infill look better. Maybe it's a cooling problem. The printing speed is similar between the two slicers?
For the 4,5, you have to check 'more perimeter -> on odd layers)

JustinCaseOfWolf, 2020-11-03:
any way to have the fan constant on at a speed?prusaslicer has
[image]

yschroeder, 2020-11-03:
Could this be related?

https://help.prusa3d.com/en/article/the-benchy-hull-line_124745

JustinCaseOfWolf, 2020-11-03:
[image]
it may as well be...

JustinCaseOfWolf, 2020-11-03:
[image]
20% constant fan fixes lots of the issues however there is still one wall line (does not go around the print) which is the problem ive noticed on prusa slicer 
[image]

JustinCaseOfWolf, 2020-11-03:
[image]

yschroeder, 2020-11-03:
Maybe it was not explicit enough, but basically, this was what I suspected in my very first post here: Different part cooling results in different extrusion characteristics, which then produces visibles lines.

As I said, your layer lines coincided with cooling fan percentage.

That very last visible line coincides with the large infill layer. That might very well be exactly the problem described in the Prusa article about the Benchy Hull Line.

So my guess is, there is nothing to do here for SuperSlicer?

JustinCaseOfWolf, 2020-11-03:
[image]

JustinCaseOfWolf, 2020-11-03:
> Maybe it was not explicit enough, but basically, this was what I suspected in my very first post here: Different part cooling results in different extrusion characteristics, which then produces [...]

[image]
one is kisslicer one is superslicer.

JustinCaseOfWolf, 2020-11-03:
[image]

JustinCaseOfWolf, 2020-11-03:
[image]

supermerill, 2020-11-03:
> So my guess is, there is nothing to do here for SuperSlicer?

I have to do something about cooling to avoid these problems.
for example, having the short layer time kicking in hard for only 1-2 layers isn't good. It should be a much more smooth increase over layers and just wait on infill if it's really too short.

JustinCaseOfWolf, 2020-11-03:
if you can picture the shape as a square then the first side is bulged towards outside - i would expect too much cooling to make the plastic shrink towards inside. would it be of any help to send you the gcode for above 2 prints?

supermerill, 2020-11-03:
no, thanks. I have to create (or someone else) the smallest and simplest piece that reproduce the bad behavior, to have less lines to read & compare.

JustinCaseOfWolf, 2020-11-03:
natural light to the rescue :
[image]

JustinCaseOfWolf, 2020-11-03:
[z_tensioner_bracket_b_x2 cut.zip](https://github.com/supermerill/SuperSlicer/files/5482217/z_tensioner_bracket_b_x2.cut.zip)
this is the smaller section - 13 minute print

JustinCaseOfWolf, 2020-11-03:
secondary problem: after the outside tab finishes the print starts on one corner which makes it round and bulged:

[image]

kisslicer doesnt have any visible seams on anywhere on the surface

supermerill, 2020-11-03:
the seam thing should be fixed on 2.3.55, hopefully

JustinCaseOfWolf, 2020-11-03:
i am looking at the gcode around where the bulge is and i can see the extruder value is bigger around there, tho i dont know how much it matters.
0.14319 (7.0) / 0.13566 (7.2) / 0.13116 (7.4) / ... / 0.12705 (8.2)
i will try to extract all the lines

JustinCaseOfWolf, 2020-11-03:
7.0:
76   G1 X101.711 Y133.609 E0.14319
178  G1 X101.694 Y134.006 E0.15564
298  G1 X101.676 Y134.403 E0.16809
437  G1 X101.659 Y134.790 E0.17115

76   G1 X101.680 Y133.609 E0.13566
175  G1 X101.672 Y134.006 E0.14785
293  G1 X101.663 Y134.403 E0.16005
425  G1 X101.655 Y134.790 E0.16327

7.4:
73   G1 X101.680 Y133.609 E0.13116
171  G1 X101.672 Y134.006 E0.14335
287  G1 X101.663 Y134.403 E0.15555
419  G1 X101.655 Y134.790 E0.15900

7.6:

7.8:

8.0:

8.2:
77   G1 X101.680 Y133.609 E0.12705
171  G1 X101.672 Y134.006 E0.13924
286  G1 X101.663 Y134.403 E0.15143
412  G1 X101.655 Y134.790 E0.15509

supermerill, 2020-11-03:
what is the first number?
` G1 X101.680 Y133.609 E0.12705`  say to go from current position to `X101.680 Y133.609` and extruding `0.12705` at the same time... but we need the current position to see if the flow is identical.

btw, it's better to create a post-process test to see if the perimeter flow is ok. But it's exactly the use of the preview panel, select the width, and see if something is wrong.

JustinCaseOfWolf, 2020-11-03:
[image]
i am trying to extract the lines red - give me a bit of time and i will get them updated above

JustinCaseOfWolf, 2020-11-03:
preview panel looks good on width, nothing odd around that layer(s)

JustinCaseOfWolf, 2020-11-03:
i thought i will do the right tab as reference / example. 

i did the left tab on both superslicer gcode and kisslicer gcode.

i did include the 6.8 layer (last full layer) and ive noticed that one looks very different to kisslicer. i am still crunching numbers but here is what i have so far. the format is layer number followed by 4 perimeters in format from-x, from-y, to-x, to-y, extruder advance

reference (right tab) in superslicer - this does NOT show the line:

6.8:
X143.289 Y133.609 X139.470 Y133.609 E0.11476
X143.306 Y134.006 X139.073 Y134.006 E0.12721
X143.324 Y134.403 X138.676 Y134.403 E0.13966
X143.341 Y134.790 X101.412 Y134.779 E1.19634

7.0:
X143.289 Y133.609 X138.524 Y133.609 E0.14318
X143.306 Y134.006 X138.127 Y134.006 E0.15563
X143.324 Y134.403 X137.729 Y134.403 E0.16809
X143.341 Y134.790 X137.342 Y134.790 E0.17115

7.2:
X143.289 Y133.609 X138.805 Y133.609 E0.13474
X143.306 Y134.006 X138.407 Y134.006 E0.14719
X143.324 Y134.403 X138.010 Y134.403 E0.15965
X143.341 Y134.790 X137.623 Y134.790 E0.16313

7.4:
X143.289 Y133.609 X138.954 Y133.609 E0.13024
X143.306 Y134.006 X138.557 Y134.006 E0.14269
X143.324 Y134.403 X138.160 Y134.403 E0.15515
X143.341 Y134.790 X137.773 Y134.790 E0.15886

7.6:
X143.289 Y133.609 X139.045 Y133.609 E0.12753
X143.306 Y134.006 X138.648 Y134.006 E0.13998
X143.324 Y134.403 X138.250 Y134.403 E0.15243
X143.341 Y134.790 X137.863 Y134.790 E0.15628

7.8:
X143.289 Y133.609 X139.084 Y133.609 E0.12635
X143.306 Y134.006 X138.687 Y134.006 E0.13880
X143.324 Y134.403 X138.290 Y134.403 E0.15126
X143.341 Y134.790 X137.902 Y134.790 E0.15517

8.0:
X143.289 Y133.609 X139.091 Y133.609 E0.12613
X143.306 Y134.006 X138.694 Y134.006 E0.13858
X143.324 Y134.403 X138.297 Y134.403 E0.15103
X143.341 Y134.790 X137.910 Y134.790 E0.15495

8.2:
X143.289 Y133.609 X139.091 Y133.609 E0.12613
X143.306 Y134.006 X138.694 Y134.006 E0.13858
X143.324 Y134.403 X138.297 Y134.403 E0.15103
X143.341 Y134.790 X137.910 Y134.790 E0.15495

the left tab which shows the bulge line is:

6.8:
X105.530 Y133.609 X101.680 Y133.609 E0.11568
X105.928 Y134.006 X101.672 Y134.006 E0.12787
X106.325 Y134.403 X101.663 Y134.403 E0.14006
X143.341 Y134.790 X101.412 Y134.779 E1.19634

7.0:
X106.476 Y133.609 X101.711 Y133.609 E0.14319
X106.874 Y134.006 X101.694 Y134.006 E0.15564
X107.271 Y134.403 X101.676 Y134.403 E0.16809
X107.658 Y134.790 X101.659 Y134.790 E0.17115

7.2 :
X106.196 Y133.609 X101.680 Y133.609 E0.13566
X106.593 Y134.006 X101.672 Y134.006 E0.14785
X106.990 Y134.403 X101.663 Y134.403 E0.16005
X107.377 Y134.790 X101.655 Y134.790 E0.16327

7.4:
X106.046 Y133.609 X101.680 Y133.609 E0.13116
X106.443 Y134.006 X101.672 Y134.006 E0.14335
X106.840 Y134.403 X101.663 Y134.403 E0.15555
X107.227 Y134.790 X101.655 Y134.790 E0.15900

7.6:
X105.956 Y133.609 X101.680 Y133.609 E0.12845
X106.353 Y134.006 X101.672 Y134.006 E0.14064
X106.750 Y134.403 X101.663 Y134.403 E0.15283
X107.137 Y134.790 X101.655 Y134.790 E0.15642

7.8:
X105.916 Y133.609 X101.680 Y133.609 E0.12727
X106.313 Y134.006 X101.672 Y134.006 E0.13946
X106.711 Y134.403 X101.663 Y134.403 E0.15166
X107.098 Y134.790 X101.655 Y134.790 E0.15531

8.0:
X105.909 Y133.609 X101.680 Y133.609 E0.12705
X106.306 Y134.006 X101.672 Y134.006 E0.13924
X106.703 Y134.403 X101.663 Y134.403 E0.15143
X107.090 Y134.790 X101.655 Y134.790 E0.15509

8.2:
X105.909 Y133.609 X101.680 Y133.609 E0.12705
X106.306 Y134.006 X101.672 Y134.006 E0.13924
X106.703 Y134.403 X101.663 Y134.403 E0.15143
X107.090 Y134.790 X101.655 Y134.790 E0.15509

kisslicer is:
6.8:
X137.001 Y124.382 X101.599 Y124.38 E1.0715
X143.365 Y124.781 X101.663 Y124.782 E1.26219
X137.132 Y123.982 X101.568 Y123.978 E1.097928
X107.229 Y123.582 X101.468 Y123.572 E0.177878

7.0:
X107.174 Y124.382 X101.444 Y124.374 E0.17341
X107.579 Y124.783 X101.422 Y124.773 E0.18637
X106.842 Y123.983 X101.456 Y123.972 E0.16628
X106.443 Y123.583 X101.468 Y123.572 E0.153592

7.2:
X106.829 Y124.382 X101.444 Y124.374 E0.16298
X107.236 Y124.783 X101.422 Y124.773 E0.17598
X106.391 Y123.983 X101.456 Y123.972 E0.152337
X106.026 Y123.583 X101.467 Y123.572 E0.14074

7.4:
X106.754 Y124.382 X101.693 Y124.382 E0.15318
X106.586 Y124.783 X101.422 Y124.773 E0.15632
X105.866 Y123.982 X101.456 Y123.972 E0.13617
X105.54 Y123.582 X101.467 Y123.572 E0.125735

7.6:
X106.618 Y124.382 X101.444 Y124.374 E0.15659
X107.021 Y124.783 X101.422 Y124.773 E0.16948
X106.259 Y123.983 X101.456 Y123.972 E0.148277
X105.928 Y123.583 X101.467 Y123.572 E0.13769

7.8:
X106.62 Y124.382 X101.444 Y124.374 E0.15666
X107.009 Y124.783 X101.422 Y124.773 E0.16911
X106.289 Y123.983 X101.456 Y123.972 E0.149206
X105.889 Y123.583 X101.467 Y123.572 E0.136496

8.0:
X106.61 Y124.382 X101.444 Y124.374 E0.15635
X107.011 Y124.783 X101.422 Y124.773 E0.16917
X106.281 Y123.983 X101.456 Y123.972 E0.148961
X105.881 Y123.583 X101.467 Y123.572 E0.136262

8.2:
X106.61 Y124.382 X101.444 Y124.374 E0.15635
X107.011 Y124.783 X101.422 Y124.773 E0.16917
X106.281 Y123.983 X101.456 Y123.972 E0.148961
X105.881 Y123.583 X101.467 Y123.572 E0.136262

JustinCaseOfWolf, 2020-11-03:
looks like it is a red herring :(

reference (right - no lines) 
Column1.1	Column1.2	Column1.3	Column1.4	Column1.5	Column1.6	Column1.7
6.8:						
143.289	133.609	139.47	133.609	0.11476	3.819	0.03005
143.306	134.006	139.073	134.006	0.12721	4.233	0.030052
143.324	134.403	138.676	134.403	0.13966	4.648	0.030047
143.341	134.79	101.412	134.779	1.19634	41.929	0.028533
7.0:						
143.289	133.609	138.524	133.609	0.14318	4.765	0.030048
143.306	134.006	138.127	134.006	0.15563	5.179	0.03005
143.324	134.403	137.729	134.403	0.16809	5.595	0.030043
143.341	134.79	137.342	134.79	0.17115	5.999	0.02853
7.2:						
143.289	133.609	138.805	133.609	0.13474	4.484	0.030049
143.306	134.006	138.407	134.006	0.14719	4.899	0.030045
143.324	134.403	138.01	134.403	0.15965	5.314	0.030043
143.341	134.79	137.623	134.79	0.16313	5.718	0.028529
7.4:						
143.289	133.609	138.954	133.609	0.13024	4.335	0.030044
143.306	134.006	138.557	134.006	0.14269	4.749	0.030046
143.324	134.403	138.16	134.403	0.15515	5.164	0.030045
143.341	134.79	137.773	134.79	0.15886	5.568	0.028531
7.6:						
143.289	133.609	139.045	133.609	0.12753	4.244	0.030049
143.306	134.006	138.648	134.006	0.13998	4.658	0.030052
143.324	134.403	138.25	134.403	0.15243	5.074	0.030041
143.341	134.79	137.863	134.79	0.15628	5.478	0.028529
7.8:						
143.289	133.609	139.084	133.609	0.12635	4.205	0.030048
143.306	134.006	138.687	134.006	0.1388	4.619	0.03005
143.324	134.403	138.29	134.403	0.15126	5.034	0.030048
143.341	134.79	137.902	134.79	0.15517	5.439	0.028529
8.0:						
143.289	133.609	139.091	133.609	0.12613	4.198	0.030045
143.306	134.006	138.694	134.006	0.13858	4.612	0.030048
143.324	134.403	138.297	134.403	0.15103	5.027	0.030044
143.341	134.79	137.91	134.79	0.15495	5.431	0.028531
8.2:						
143.289	133.609	139.091	133.609	0.12613	4.198	0.030045
143.306	134.006	138.694	134.006	0.13858	4.612	0.030048
143.324	134.403	138.297	134.403	0.15103	5.027	0.030044
143.341	134.79	137.91	134.79	0.15495	5.431	0.028531

left - bulging
6.8:						
137.001	124.382	101.599	124.38	1.0715	35.402	0.030267
143.365	124.781	101.663	124.782	1.26219	41.702	0.030267
137.132	123.982	101.568	123.978	1.097928	35.564	0.030872
107.229	123.582	101.468	123.572	0.177878	5.761009	0.030876
7.0:						
107.174	124.382	101.444	124.374	0.17341	5.730006	0.030263
107.579	124.783	101.422	124.773	0.18637	6.157008	0.03027
106.842	123.983	101.456	123.972	0.16628	5.386011	0.030873
106.443	123.583	101.468	123.572	0.153592	4.975012	0.030873
7.2:						
106.829	124.382	101.444	124.374	0.16298	5.385006	0.030266
107.236	124.783	101.422	124.773	0.17598	5.814009	0.030268
106.391	123.983	101.456	123.972	0.152337	4.935012	0.030869
106.026	123.583	101.467	123.572	0.14074	4.559013	0.030871
7.4:						
106.754	124.382	101.693	124.382	0.15318	5.061	0.030267
106.586	124.783	101.422	124.773	0.15632	5.16401	0.030271
105.866	123.982	101.456	123.972	0.13617	4.410011	0.030877
105.54	123.582	101.467	123.572	0.125735	4.073012	0.03087
7.6:						
106.618	124.382	101.444	124.374	0.15659	5.174006	0.030265
107.021	124.783	101.422	124.773	0.16948	5.599009	0.03027
106.259	123.983	101.456	123.972	0.148277	4.803013	0.030872
105.928	123.583	101.467	123.572	0.13769	4.461014	0.030865
7.8:						
106.62	124.382	101.444	124.374	0.15666	5.176006	0.030267
107.009	124.783	101.422	124.773	0.16911	5.587009	0.030268
106.289	123.983	101.456	123.972	0.149206	4.833013	0.030872
105.889	123.583	101.467	123.572	0.136496	4.422014	0.030867
8.0:						
106.61	124.382	101.444	124.374	0.15635	5.166006	0.030265
107.011	124.783	101.422	124.773	0.16917	5.589009	0.030268
106.281	123.983	101.456	123.972	0.148961	4.825013	0.030873
105.881	123.583	101.467	123.572	0.136262	4.414014	0.03087
8.2:						
106.61	124.382	101.444	124.374	0.15635	5.166006	0.030265
107.011	124.783	101.422	124.773	0.16917	5.589009	0.030268
106.281	123.983	101.456	123.972	0.148961	4.825013	0.030873
105.881	123.583	101.467	123.572	0.136262	4.414014	0.03087

kisslicer
6.8:						
137.001	124.382	101.599	124.38	1.0715	35.402	0.030267
143.365	124.781	101.663	124.782	1.26219	41.702	0.030267
137.132	123.982	101.568	123.978	1.097928	35.564	0.030872
107.229	123.582	101.468	123.572	0.177878	5.761009	0.030876
7.0:						
107.174	124.382	101.444	124.374	0.17341	5.730006	0.030263
107.579	124.783	101.422	124.773	0.18637	6.157008	0.03027
106.842	123.983	101.456	123.972	0.16628	5.386011	0.030873
106.443	123.583	101.468	123.572	0.153592	4.975012	0.030873
7.2:						
106.829	124.382	101.444	124.374	0.16298	5.385006	0.030266
107.236	124.783	101.422	124.773	0.17598	5.814009	0.030268
106.391	123.983	101.456	123.972	0.152337	4.935012	0.030869
106.026	123.583	101.467	123.572	0.14074	4.559013	0.030871
7.4:						
106.754	124.382	101.693	124.382	0.15318	5.061	0.030267
106.586	124.783	101.422	124.773	0.15632	5.16401	0.030271
105.866	123.982	101.456	123.972	0.13617	4.410011	0.030877
105.54	123.582	101.467	123.572	0.125735	4.073012	0.03087
7.6:						
106.618	124.382	101.444	124.374	0.15659	5.174006	0.030265
107.021	124.783	101.422	124.773	0.16948	5.599009	0.03027
106.259	123.983	101.456	123.972	0.148277	4.803013	0.030872
105.928	123.583	101.467	123.572	0.13769	4.461014	0.030865
7.8:						
106.62	124.382	101.444	124.374	0.15666	5.176006	0.030267
107.009	124.783	101.422	124.773	0.16911	5.587009	0.030268
106.289	123.983	101.456	123.972	0.149206	4.833013	0.030872
105.889	123.583	101.467	123.572	0.136496	4.422014	0.030867
8.0:						
106.61	124.382	101.444	124.374	0.15635	5.166006	0.030265
107.011	124.783	101.422	124.773	0.16917	5.589009	0.030268
106.281	123.983	101.456	123.972	0.148961	4.825013	0.030873
105.881	123.583	101.467	123.572	0.136262	4.414014	0.03087
8.2:						
106.61	124.382	101.444	124.374	0.15635	5.166006	0.030265
107.011	124.783	101.422	124.773	0.16917	5.589009	0.030268
106.281	123.983	101.456	123.972	0.148961	4.825013	0.030873
105.881	123.583	101.467	123.572	0.136262	4.414014	0.03087

JustinCaseOfWolf, 2020-11-04:
quick update i did try printing the sliced stl from above and while it still has a line it is much smaller than in the original print. i think i have messed up my configuration :)

vospascal, 2024-01-04:
@JustinCaseOfWolf if its solved then you should close the thread

JustinCaseOfWolf, 2024-01-08:
@vospascal  what makes you think it is solved? as far as i am aware no work has been done on this issue.