supermerill/SuperSlicer #619: Error with updated translations

boromyr, 2020-10-31:
### Version
2.2.53.3 and later

### Operating system type + version
Windows 10 64bit

### Behavior
I have followed the procedure to update the new translations, but I continue to receive the errors shown in the photos.
I created the it.po file with pomergeur.jar, then with Poedit I updated the translations and compiled the .mo file, which I further checked using a .mo editor, but without success.
I also tried several online sites to compile the .mo file, as well as repeating all steps from 3 weeks to today again.

[image]
[image]

supermerill, 2020-10-31:
I think there is a missing (or too much)  %f or %something in a line.
Some mo file editors can detect these kind of problems.

last time that happens to me, I use a dichotimic search:
* i erased the last half of the .mo, compile and test it
  * if it don't work, repeat again
  * if it works, add the half of the erased section
* when the erased section is small enough, try to spot the problem, or if nothing wrong is seen, continue to erase/add until you have the problematic translation.

I'll see if I can output a more useful error message in the meantime.

boromyr, 2020-11-01:
Perfect thank you very much! I managed to isolate the problem: anyone who wants to update translations must search and replace entries like "% 1 %" or "% 3 %%%" with the same ones but without spaces.
Can I send you the updated translations in Italian, so you can add them in future releases to make them available to everyone?

boromyr, 2020-11-01:
Traduzioni aggiornate in italiano(traduttore automatico - non complete al 100% - in futuro aggiornamento)
Updated translations in Italian (automatic translator - not 100% complete - in the future update)
[SuperSlicer_it.zip](https://github.com/supermerill/SuperSlicer/files/5470442/SuperSlicer_it.zip)

boromyr, 2020-11-23:
Italian translations updated for version 2.2.55.1
Translations via Bing and Google Translate, I have also started to replace some "mechanical" translations with "human" translations.
Replace the files in "... \ SuperSlicer \ resources \ localization \ it"
[it.zip](https://github.com/supermerill/SuperSlicer/files/5583009/it.zip)

@supermerill I found many entries not present in the .pot and .po files, in fact although I have translated all the sentences in the .po files many remain untranslated. I guess the files are yet to be updated.