supermerill/SuperSlicer #620: Vase mode on demand

espr14, 2020-11-01:
### Version
v2.2.54

### Operating system type + version
any

### 3D printer brand / version + firmware version (if known)
any

### Behavior
Standard vase mode requires 1 perimeter, no top infill etc. That's fine for vases but not for many other models. Some models need something on top and more perimeters on some places like in the picture below (horizontal orientation but it is printed vertically).
I propose vase mode on demand. This would not need any restriction. The vase mode would be used only on places where single perimeter, no infill and no other object is printed.
Bonus: in case of more perimeters and no infill it could offer merging perimeters into one or omitting inner perimeters.

[image]

philipphirte, 2022-09-11:
Is there a timeline on when this could be implemented?

supermerill, 2022-09-12:
no timeline, many things may be more critical in the near term.

Only to have a timeline is if someone makes it (or pay me to be able to dedicate more time on it).