supermerill/SuperSlicer #621: Compilation failure - it seems to think it's missing a file

rkarlsba, 2020-11-01:
### Version
From git, Date:   Fri Oct 30 22:45:30 2020 +0100

### Operating system type + version
Darwin Roys-MacBook-Pro-2529.local 19.4.0 Darwin Kernel Version 19.4.0: Wed Mar  4 22:28:40 PST 2020; root:xnu-6153.101.6~15/RELEASE_X86_64 x86_64 i386 MacBookPro11,5 Darwin

### 3D printer brand / version + firmware version (if known)
Irrelevant for now

### Behavior
Roys-MacBook-Pro-2529:build roy$ cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_ENC_CHECK: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: /Users/roy/src/git/SuperSlicer/build/../deps/build/destdir/usr/local (from cache or command line)
OS X SDK Path: /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk
OS X Deployment Target: (default)
-- Found Boost: /usr/local/lib/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.64.0") found components: system filesystem thread log locale regex chrono atomic date_time
-- Boost::boost exists
--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = /Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/include
--   TBB_DEFINITIONS         =
--   TBB_LIBRARIES           = /Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libtbb_static.a
--   TBB_DEFINITIONS_DEBUG   = TBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     =
--   TBB_DEFINITIONS_RELEASE =
--   TBB_LIBRARIES_RELEASE   = /Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libtbb_static.a
-- FindGLEW: did not find GLEW CMake config file. Searching for libraries.
-- FindGLEW: Found OpenGL Framework.
-- FindGLEW: OPENGL_LIBRARIES: /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/OpenGL.framework;/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/OpenGL.framework
-- FindGLEW: GLEW_USE_STATIC_LIBS: ON.
-- FindGLEW: GLEW_INCLUDE_DIR: /Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/include
-- FindGLEW: GLEW_INCLUDE_DIRS: /Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/include
-- FindGLEW: CMAKE_FIND_LIBRARY_SUFFIXES for SHARED: .tbd;.dylib;.so;.a
-- FindGLEW: CMAKE_FIND_LIBRARY_SUFFIXES for STATIC: .dylib;.so
-- FindGLEW: GLEW_SHARED_LIBRARY_RELEASE: GLEW_SHARED_LIBRARY_RELEASE-NOTFOUND
-- FindGLEW: GLEW_STATIC_LIBRARY_RELEASE: /Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libglew.a
-- FindGLEW: GLEW_SHARED_LIBRARY_DEBUG: GLEW_SHARED_LIBRARY_DEBUG-NOTFOUND
-- FindGLEW: GLEW_STATIC_LIBRARY_DEBUG: GLEW_STATIC_LIBRARY_DEBUG-NOTFOUND
-- FindGLEW: GLEW_SHARED_LIBRARY: GLEW_SHARED_LIBRARY-NOTFOUND
-- FindGLEW: GLEW_STATIC_LIBRARY: /Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libglew.a
-- FindGLEW: GLEW_LIBRARIES: /Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libglew.a
-- FindGLEW: GLEW_VERSION_MAJOR: 1
-- FindGLEW: GLEW_VERSION_MINOR: 13
-- FindGLEW: GLEW_VERSION_MICRO: 0
-- FindGLEW: GLEW_VERSION: 1.13.0
-- FindGLEW: Creating GLEW::glew_s imported target.
-- FindGLEW: Creating GLEW::GLEW imported target.
-- Found NLopt in '/Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib'.
-- Using NLopt include directory '/Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/include'.
-- Using NLopt library '/Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libnlopt.a'.
-- OpenVDB ABI Version: 6
-- Found Boost: /usr/local/lib/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: iostreams system
-- OpenVDB libraries: /Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libopenvdb.a
-- Miniz NOT found in system, using bundled version...
-- Using qhull from system.
-- IGL NOT found, using bundled version...
-- Using header-only CGAL
-- Targetting Unix Makefiles
-- Using /Library/Developer/CommandLineTools/usr/bin/c++ compiler.
-- DARWIN_VERSION=19
-- Mac Leopard detected
-- Found Boost: /usr/local/lib/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.48")
-- Boost include dirs: /usr/local/include
-- Boost libraries:
-- wx-config path: /Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/bin/wx-config
-- wx libs: -L/Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib;;;-framework IOKit;-framework Carbon;-framework Cocoa;-framework AudioToolbox;-framework System;-framework OpenGL;/Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_gl-3.1.a;/Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_stc-3.1.a;/Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_scintilla-3.1.a;/Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_html-3.1.a;/Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_core-3.1.a;/Users/roy/src/git/SuperSlicer/deps/build/destdir/usr/local/lib/libwx_baseu-3.1.a;-framework OpenGL;-framework AGL;-framework WebKit;-lwxtiff-3.1;-lwxjpeg-3.1;-lwxpng-3.1;-lwxregexu-3.1;-lwxscintilla-3.1;-lwxexpat-3.1;-lz;-framework Security;-lpthread;-liconv
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/roy/src/git/SuperSlicer/build
Roys-MacBook-Pro-2529:build roy$ make
[  1%] Built target expat
[  2%] Built target hidapi
[  2%] Built target encoding-check
[  3%] Checking source files encodings for target libslic3r_gui
[  3%] Built target encoding-check-libslic3r_gui
[  3%] Checking source files encodings for target avrdude-slic3r
[  3%] Built target encoding-check-avrdude-slic3r
[  4%] Checking source files encodings for target avrdude
[  4%] Built target encoding-check-avrdude
[  4%] Checking source files encodings for target semver
[  4%] Built target encoding-check-semver
[  4%] Checking source files encodings for target libslic3r
[  4%] Built target encoding-check-libslic3r
[  4%] Built target global-encoding-check
[  4%] Building CXX object src/admesh/CMakeFiles/admesh.dir/stlinit.cpp.o
/Users/roy/src/git/SuperSlicer/src/admesh/stlinit.cpp:31:10: fatal error: 'boost/detail/endian.hpp' file not found
#include <boost/detail/endian.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [src/admesh/CMakeFiles/admesh.dir/stlinit.cpp.o] Error 1
make[1]: *** [src/admesh/CMakeFiles/admesh.dir/all] Error 2
make: *** [all] Error 2
Roys-MacBook-Pro-2529:build roy$

#### Project File (.3MF) where problem occurs

supermerill, 2020-11-01:
the nightly build can build
https://github.com/supermerill/SuperSlicer/runs/1334596892?check_suite_focus=true
I'll re-run it this night, to be sure.

For your errir, i'll try by replacing "$PWD/../deps/build/destdir/usr/local" by a full path, not one that need to be executed.

Lukas1818, 2020-12-31:
I get the same issue on Arch LInux with boost 1.75 and SuperSlicer 2.2.54.2. But SuperSlicer 2.2.54.2, Boost 1.71.0 and SuperSlicer master 996a42f2f95117805470b62f1504a86ae6ca33a0, Boost 1.75 does compile.

Lukas1818, 2020-12-31:
It is a breaking change of boost 1.73. The commit 3a194f91871684b5fc8c6c78300b87e0ee1cb610 fix the changes.

Using 3a194f91871684b5fc8c6c78300b87e0ee1cb610 as patch allow you to compile SuperSlicer 2.2.54.2 with boost 1.73 and newer. (I tested this only with boost 1.75)

closed 2021-06-26 by supermerill