supermerill/SuperSlicer #622: ABRT when setting a platter texture

lovesegfault, 2020-11-01:
### Version
2.2.54

### Operating system type + version
NixOS 20.09

### 3D printer brand / version + firmware version (if known)
Voron V0

### Behavior
When starting with the default profile and applying a png texture to the platter the program aborts. Here's the backtrace:
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
        set = {__val = {0, 139868025922880, 139867891701232, 139867891701520, 139867891701267, 139870422610136, 139868025922880, 139870422727284, 36, 139870422728251, 139868025922880, 0, 139870422912999, 139867891701268, 139867891701233, 0}}
        pid = <optimized out>
        tid = <optimized out>
        ret = <optimized out>
#1  0x00007f36222dc523 in __GI_abort () at abort.c:79
        save_stage = 1
        act = {__sigaction_handler = {sa_handler = 0x8000000000000000, sa_sigaction = 0x8000000000000000}, sa_mask = {__val = {9223372036854775808, 13835058058503389184, 4611686019501129728, 147502061883285, 109899623200333, 147502061883285, 109899623200333, 147502061883285, 109899623200333, 147502061883285, 109899623200333, 76125088, 14766316007815423488, 139865610125314, 139868344661792, 139868344662048}}, sa_flags = -1560299744, sa_restorer = 0x1000}
        sigs = {__val = {32, 139870422512237, 206158430256, 0, 139868344661536, 139867891733824, 139870040880618, 0, 30, 161048388698301, 144577189144256, 147502061846528, 109899623200333, 147502061883285, 109899623200333, 147502061883285}}
#2  0x00007f3622332b28 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7f362243d533 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
        ap = {{gp_offset = 24, fp_offset = 32565, overflow_arg_area = 0x7f35a2ffb830, reg_save_area = 0x7f35a2ffb7c0}}
        fd = 2
        list = <optimized out>
        nlist = <optimized out>
        cp = <optimized out>
#3  0x00007f362233a2fa in malloc_printerr (str=str@entry=0x7f362243fff8 "malloc(): invalid size (unsorted)") at malloc.c:5389
No locals.
#4  0x00007f362233d4b4 in _int_malloc (av=av@entry=0x7f3588000020, bytes=bytes@entry=100) at malloc.c:3764
        next = <optimized out>
        iters = <optimized out>
        nb = <optimized out>
        idx = 7
        bin = <optimized out>
        victim = <optimized out>
        size = <optimized out>
        victim_index = <optimized out>
        remainder = <optimized out>
        remainder_size = <optimized out>
        block = <optimized out>
        bit = <optimized out>
        map = <optimized out>
        fwd = <optimized out>
        bck = <optimized out>
        tcache_unsorted_count = 0
        tcache_nb = 112
        tc_idx = 5
        return_cached = 0
        __PRETTY_FUNCTION__ = "_int_malloc"
#5  0x00007f362233ed68 in __GI___libc_malloc (bytes=100) at malloc.c:3086
        ar_ptr = 0x7f3588000020
        victim = <optimized out>
        hook = <optimized out>
        tbytes = <optimized out>
        tc_idx = <optimized out>
        __PRETTY_FUNCTION__ = "__libc_malloc"
#6  0x00007f3624571615 in operator new(unsigned long) () from /nix/store/fym51l26jrpzhlvaj607wd3im9511j2q-gcc-9.3.0-lib/lib/libstdc++.so.6
No symbol table info available.
#7  0x00000000005c5383 in std::vector<unsigned char, std::allocator<unsigned char> >::_M_default_append(unsigned long) ()
No symbol table info available.
#8  0x0000000000e1df1c in Slic3r::GUI::GLTexture::Compressor::compress() ()
No symbol table info available.
#9  0x00007f362459a930 in ?? () from /nix/store/fym51l26jrpzhlvaj607wd3im9511j2q-gcc-9.3.0-lib/lib/libstdc++.so.6
No symbol table info available.
#10 0x00007f3622484e9e in start_thread (arg=0x7f35a2ffd640) at pthread_create.c:463
        ret = <optimized out>
        pd = 0x7f35a2ffd640
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {139868344669760, 1793814266752971542, 140728347229566, 140728347229567, 0, 75992704, -1833055839098326250, -1835027720090766570}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = 0
#11 0x00007f36223b166f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

#### Project File (.3MF) where problem occurs
Here's the png file in question:
[image]

supermerill, 2020-11-02:
Can reproduce in 2.2.54
Can't in 2.3.55, seems like prusa has already fixed it.

closed 2020-11-25 by supermerill

lovesegfault, 2020-11-25:
So, I've not been successful using this feature on 2.3.55.2. I always get a crash.
```
❯ superslicer
12:37:51: Debug: Adding duplicate image handler for 'JPEG file'
create settings  print.ui
create settings  filament.ui
create settings  sla_print.ui
create settings  sla_material.ui
create settings  printer_fff.ui
create settings  extruder.ui
malloc(): invalid size (unsorted)
zsh: IOT instruction (core dumped)  superslicer
```

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
        set = {__val = {0 <repeats 16 times>}}
        pid = <optimized out>
        tid = <optimized out>
        ret = <optimized out>
#1  0x00007f553d7e4523 in __GI_abort () at abort.c:79
        save_stage = 1
        act = {__sigaction_handler = {sa_handler = 0x8000000000000000, sa_sigaction = 0x8000000000000000}, sa_mask = {__val = {9223372036854775808, 13835058058503389184, 4611686019501129728, 147502061883285, 109899623200333, 147502061883285, 109899623200333, 147502061883285, 109899623200333, 147502061883285, 109899623200333, 0, 15144457914011885824, 140001547363936, 140001547363936, 140001547364192}}, sa_flags = -1501583776, sa_restorer = 0x1000}
        sigs = {__val = {32, 0, 0, 0, 0, 0, 0, 0, 0, 161048388698301, 144577189144256, 147502061846528, 109899623200333, 147502061883285, 109899623200333, 147502061883285}}
#2  0x00007f553d83ab28 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7f553d945533 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
        ap = {{gp_offset = 24, fp_offset = 32596, overflow_arg_area = 0x7f54a67fa770, reg_save_area = 0x7f54a67fa700}}
        fd = 2
        list = <optimized out>
        nlist = <optimized out>
        cp = <optimized out>
#3  0x00007f553d8422fa in malloc_printerr (str=str@entry=0x7f553d947ff8 "malloc(): invalid size (unsorted)") at malloc.c:5389
No locals.
#4  0x00007f553d8454b4 in _int_malloc (av=av@entry=0x7f5484000020, bytes=bytes@entry=100) at malloc.c:3764
        next = <optimized out>
        iters = <optimized out>
        nb = <optimized out>
        idx = 7
        bin = <optimized out>
        victim = <optimized out>
        size = <optimized out>
        victim_index = <optimized out>
        remainder = <optimized out>
        remainder_size = <optimized out>
        block = <optimized out>
        bit = <optimized out>
        map = <optimized out>
        fwd = <optimized out>
        bck = <optimized out>
        tcache_unsorted_count = 0
        tcache_nb = 112
        tc_idx = 5
        return_cached = 0
        __PRETTY_FUNCTION__ = "_int_malloc"
#5  0x00007f553d846d68 in __GI___libc_malloc (bytes=100) at malloc.c:3086
        ar_ptr = 0x7f5484000020
        victim = <optimized out>
        hook = <optimized out>
        tbytes = <optimized out>
        tc_idx = <optimized out>
        __PRETTY_FUNCTION__ = "__libc_malloc"
#6  0x00007f553fae5615 in operator new(unsigned long) () from /nix/store/51hq0xxp9nng3xxfz7dpkhb9lzy7sz84-gcc-9.3.0-lib/lib/libstdc++.so.6
No symbol table info available.
#7  0x00000000005eded3 in std::vector<unsigned char, std::allocator<unsigned char> >::_M_default_append(unsigned long) ()
No symbol table info available.
#8  0x0000000000f3ff1c in Slic3r::GUI::GLTexture::Compressor::compress() ()
No symbol table info available.
#9  0x00007f553fb0e930 in ?? () from /nix/store/51hq0xxp9nng3xxfz7dpkhb9lzy7sz84-gcc-9.3.0-lib/lib/libstdc++.so.6
No symbol table info available.
#10 0x00007f553d98ae9e in start_thread (arg=0x7f54a67fc640) at pthread_create.c:463
        ret = <optimized out>
        pd = 0x7f54a67fc640
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140001547372096, -2825782087285915057, 140722505427150, 140722505427151, 0, 76075584, 2765635504382231119, 2765972238367801935}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = 0
#11 0x00007f553d8b966f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```