supermerill/SuperSlicer #626: Filament specific custom start gcode is not included

Matts-Hub, 2020-11-03:
### Version
Version 2.2.53

### Operating system type + version
Windows 10 x64

### Behavior
Start G-code entered in the filament settings is not included in the exported gcode file. The end gcode is though. You can see in the attached project file, I am setting my linear advance K factor in the filament settings, but the code is not exported.

#### Project File (.3MF) where problem occurs
[demo.zip](https://github.com/supermerill/SuperSlicer/files/5479162/demo.zip)

neophyl, 2020-11-03:
I think this is a duplicate of https://github.com/supermerill/SuperSlicer/issues/575  ?  That has it listed as fixed in last release, which would be 2.2.54

Matts-Hub, 2020-11-03:
Thank you, I tried searching but didn't find that one. That issue is exactly what I'm talking about, I'll try the latest release!

closed 2020-11-03 by Matts-Hub